nasa/osal #6: Test `bin-sem-flush-test` does not compile

ChristopherRabotin, 2017-12-14:
OSAL fails to compile with a clean clone of the repo.

```
(.venv) chris@linux-fkyc:~/Workspace/osal> source setvars.sh 
(.venv) chris@linux-fkyc:~/Workspace/osal> cd build/
(.venv) chris@linux-fkyc:~/Workspace/osal/build> make config
test -d inc || mkdir inc
cp -f ../src/bsp/pc-linux/config/osconfig.h inc/osconfig.h
(.venv) chris@linux-fkyc:~/Workspace/osal/build> make
make -C tests 
make[1]: Entering directory '/home/chris/Workspace/osal/build/tests'
make -C bin-sem-flush-test 
make[2]: Entering directory '/home/chris/Workspace/osal/build/tests/bin-sem-flush-test'
gcc  -I/home/chris/Workspace/osal/src/inc -I/home/chris/Workspace/osal/src/os/inc -I/home/chris/Workspace/osal/src/tests/bin-sem-flush-test -I/home/chris/Workspace/osal/src/../ut_assert/inc -I../../inc   -D_EL -DENDIAN=_EL -DSOFTWARE_LITTLE_BIT_ORDER  -D__ix86__ -D_ix86_ -D_LINUX_OS_ -Dposix -DX86PC -DBUILD= -D_REENTRANT -D _EMBED_  -D_XOPEN_SOURCE=600   -Wall -pedantic -std=c99 -Wstrict-prototypes -O1 -DOS_DEBUG_LEVEL=3 -c -o bin-sem-flush-test.o /home/chris/Workspace/osal/src/tests/bin-sem-flush-test/bin-sem-flush-test.c
In file included from /home/chris/Workspace/osal/src/tests/bin-sem-flush-test/bin-sem-flush-test.c:5:0:
/home/chris/Workspace/osal/src/os/inc/common_types.h:326:38: error: size of array ‘TypeUint32WrongSize’ is negative
 CompileTimeAssert(sizeof(uint32)==4, TypeUint32WrongSize);
                                      ^
/home/chris/Workspace/osal/src/os/inc/common_types.h:104:60: note: in definition of macro ‘CompileTimeAssert’
 #define CompileTimeAssert(Condition, Message) typedef char Message[(Condition) ? 1 : -1]
                                                            ^
/home/chris/Workspace/osal/src/os/inc/common_types.h:330:38: error: size of array ‘Typeint32WrongSize’ is negative
 CompileTimeAssert(sizeof(int32)==4,  Typeint32WrongSize);
                                      ^
/home/chris/Workspace/osal/src/os/inc/common_types.h:104:60: note: in definition of macro ‘CompileTimeAssert’
 #define CompileTimeAssert(Condition, Message) typedef char Message[(Condition) ? 1 : -1]
                                                            ^
/home/chris/Workspace/osal/src/make/app-rules.mak:21: recipe for target 'bin-sem-flush-test.o' failed
make[2]: *** [bin-sem-flush-test.o] Error 1
make[2]: Leaving directory '/home/chris/Workspace/osal/build/tests/bin-sem-flush-test'
Makefile:2: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/chris/Workspace/osal/build/tests'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
(.venv) chris@linux-fkyc:~/Workspace/osal/build> 
```

nickcounts, 2018-01-10:
I had a similar issue on Ubuntu.
Setting `OSAL_M32 = -m32` in `osal/build/osal-config.mak` and installing `gcc-multilib` got me to a successful build.

aogrcs, 2018-01-30:
Hi, Nick
It did works, thanks for you reply.
@nickcounts

closed 2019-09-03 by skliper