nasa/osal #11: Return OS_ERROR if OS_TimerSet is called with both parameters as zero

skliper, 2019-09-13:
**Is your feature request related to a problem? Please describe.**
Related to the discussion in #9

**Describe the solution you'd like**
See title, as described in https://github.com/nasa/osal/issues/9#issuecomment-528071853

**Describe alternatives you've considered**
See #9 comment thread

**Additional context**
None

**Requester Info**
Jacob Hageman/NASA-GSFC

avan989, 2019-09-16:
I'll do this.

avan989, 2019-09-17:
I need a little more clarification on this. There are 3 files that contains OS_TimerSet. Is the fix asking to change in all 3 (psix, rtems, vxworks6) or just in rtems?

jphickey, 2019-09-17:
Be sure to do a "pull" first.  The merge from rc-5.0.0 happened recently.  You should find just one implementation of this function in the "shared" directory.

commit 58cf9773 (2019-10-01): Merge pull request #19 from avan989/OS_ERROR

Fix #11, Update OS_TimerSet to return OS_ERROR
commit ef6863f0 (2019-10-02): Fix #11, Update OS_TimerSet to return OS_ERROR
closed 2019-10-02 by skliper