nasa/osal #22: Stand-alone posix-ng coverage test fails to build

skliper, 2019-09-19:
**Describe the bug**
Posix-ng coverage tests fail to make

**To Reproduce**
Steps to reproduce the behavior:
From a clean directory with BASE_DIR=(your cFS base directory)
```
cmake -DOSALCOVERAGE_TARGET_OSTYPE='posix-ng' $BASE_DIR/osal/src/unit-test-coverage/
make
```
builds up to making ut_posix-ng_osapi with the following error:
```
[ 42%] Building C object posix-ng/modules/CMakeFiles/ut_posix-ng_osapi.dir/src/ut-osapi.c.o
cd /home/jhageman/cFS/osal_test_dir/posix-ng/modules && /usr/bin/cc  -D_UNIT_TEST_ -I/home/jhageman/cFS/cFS-GitHub/osal/src/unit-test-coverage/ut-stubs/inc/overrides -I/home/jhageman/cFS/cFS-GitHub/osal/ut_assert/inc -I/home/jhageman/cFS/cFS-GitHub/osal/src/os/inc -I/home/jhageman/cFS/cFS-GitHub/osal/src/bsp/pc-linux/config -I/home/jhageman/cFS/cFS-GitHub/osal/src/os/shared -I/home/jhageman/cFS/cFS-GitHub/osal/src/os/portable -I/home/jhageman/cFS/cFS-GitHub/osal/src/unit-test-coverage/ut-stubs/inc -I/home/jhageman/cFS/cFS-GitHub/osal/src/os/posix-ng  -D_LINUX_OS_   -Wall -Werror -pg --coverage -o CMakeFiles/ut_posix-ng_osapi.dir/src/ut-osapi.c.o   -c /home/jhageman/cFS/cFS-GitHub/osal/src/unit-test-coverage/posix-ng/modules/src/ut-osapi.c
/home/jhageman/cFS/cFS-GitHub/osal/src/unit-test-coverage/posix-ng/modules/src/ut-osapi.c:9:19: fatal error: osapi.c: No such file or directory
compilation terminated.
```

**Expected behavior**
Clean build, and execute/pass with make test

**Code snips**
Didn't investigate

**System observed on:**
 - cFS development server (VM)
 - OS: Linux 4.4.0-146-generic
 - Versions: rc-6.7.0 bundle

**Additional context**
N/A

**Reporter Info**
Jacob Hageman/NASA-GSFC

skliper, 2019-09-19:
@jphickey shared works, posix-ng looks broken

skliper, 2019-09-19:
Looks like it's from -I/home/jhageman/cFS/cFS-GitHub/osal/src/os/posix-ng, since it's now posix

skliper, 2019-09-19:
Tried `git mv posix-ng posix`, and  -DOSALCOVERAGE_TARGET_OSTYPE='posix', and it now fails with:

```
[ 42%] Building C object posix/modules/CMakeFiles/ut_posix_osapi.dir/src/ut-osapi.c.o
In file included from /home/jhageman/cFS/cFS-GitHub/osal/src/unit-test-coverage/posix/modules/src/ut-osapi.c:9:0:
/home/jhageman/cFS/cFS-GitHub/osal/src/os/posix/osapi.c: In function ‘OS_ConsoleCreate_Impl’:
/home/jhageman/cFS/cFS-GitHub/osal/src/os/posix/osapi.c:47:33: error: ‘STDOUT_FILENO’ undeclared (first use in this function)
 #define OSAL_CONSOLE_FILENO     STDOUT_FILENO
```
and others.

jphickey, 2019-09-19:
Note that previous OSAL versions only ever had a coverage test for vxworks6.  There was never a coverage test available for posix at all.  

The stuff for posix under unit-test-coverage is the start of something, but quite incomplete.  It was never a focus of testing because AFAIK the coverage test was only a requirement for VxWorks OS.

I can probably fix the immediate build issues to make it work at least into something buildable and runnable, but note this would be a new feature as it is not something that OSAL previously offered.

skliper, 2019-09-23:
Marked as an enhancement to reflect Joe's comment (new feature).  Providing coverage metrics for rtems is likely higher priority than posix, although they are both in the plan to provide the certification framework for projects that will need to show coverage statistics.

skliper, 2020-04-10:
This issue is OBE, opening a new one to implement coverage tests for posix-ng, #412

closed 2020-04-10 by skliper