nasa/osal #102: Make compiles with --std=c99 work

skliper, 2019-09-30:
OSAL has historically attempted to be strict ANSI C90, with
obvious exceptions for cases where the code needed to use a
facility that did not yet exist.

Compile with --std=c90 if you want to see these.

The default compilation we do now does not specify that the
code should be strictly conformant to any standard, so what
we get in practice is GCC's C90 with extensions.

Turning up all the warnings using
{{{
gcc -W -Wall -Wextra -pedantic
}}}
will include a large number of very useful static code
analysis checks, but will warn about the use of C99
facilities that we want to use.

What I propose to do is to run a set of trial builds
where the build script adjusts the compiler flags,
on the fly (no commits), to be:
{{{
gcc --std=c99 -O3 -g -W -Wall -Wextra -pedantic
}}}

This asks GCC to do as thorough a job as it can do to report
any of our code that deviates from `INCITS ISO/IEC 9899:1999`.

Our code does require use of facilities not specified in C99,
and these facilities should be enabled, where necessary, by
the use of the appropriate Option Feature Macros. I anticipate that
this will only require certain specific source files to have
an annotation (before the `#include` lines) that they are to
be considered to contain `XOPEN` or `POSIX` standard sources
at a specific level of those standards.

See also CFE ticket [cfs_cfe:82]

skliper, 2020-01-07:
This has apparently been fixed, at least with BUILDTYPE=release (does -O3) and the following strict flags: `-Wall -Werror -std=c99 -D_XOPEN_SOURCE=600 -pedantic -Wstrict-prototypes -Wcast-align -Wwrite-strings`

See related issues https://github.com/nasa/cFS/issues/39, https://github.com/nasa/cFE/issues/24

closed 2020-01-07 by skliper

jphickey, 2020-01-07:
Note that `-Wcast-align` is architecture-dependent.  It still does _not_ compile cleanly on MIPS/SPARC, but there are other issue tickets covering that aspect.

I would hold off on adding at least `-Wcast-align` to the default set of flags, however, due to this.