nasa/osal #116: OS_rename() doesn't first check if a file is in use

skliper, 2019-09-30:
The osfileapi.c **OS_rename()** doesn't check if a file is in use or not (in an entry in the global OS_FDTable) before it makes the system call to rename the file.

However, this is in contrast to **OS_cp(), OS_mv(), or OS_remove()** - which do check if the file is open (according to the OS_FDTable entry).  If those functions see that the file is open then they return and don't alter the filesystem.  So this behavior seems quite different for functions at affect a file.

The documentation of the OSAL API doesn't explicitly mention any behavior tied to a file that is in use by the OSAL for any of these functions (although the implementation is very clear for those other three functions.)

Discovered in #106, as part of #45 unit testing.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-05-23 15:33:17:_

Submitted a new ticket with this issue, see #238.

Replying to [comment:5 cdknight]:
> Looking at OS_cp, which does a system call to the "cp" command, this is really ugly, not realtime-friendly, and probably a security risk (if you could somehow inject a ';' into the system call, which shouldn't be difficult...Doesn't look like OS_TranslatePath catches this.)
> 
> I suggest re-writing OS_cp to do an open, a loop of read/write, and close (with the appropriate checks to ensure errors are handled.)
> 
> Also looks like OS_TranslatePath does the check for NULL and length, so the OS_{cp|mv|remove|...} should not do the same.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-05-29 12:24:58:_

CCB 5/29/2019 - Discussed, 3 options - "wontfix", make them consistent for no check, make them consistent for all check.

Action is "wontfix" and close after updating documentation.

Suggestion is to convert the API to markdown.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-03 13:47:08:_

Moved 4.3.0 tickets that won't make release to 4.4.0

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-08 15:55:03:_

This is a documentation update only, so revision number update.  See comment:7 for action.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-08 15:58:56:_

Note [changeset:b10d3e2] in `TRAC-93+215_cp_rename` was suggested but it applies to last gen and isn't applicable to current development work.

skliper, 2020-03-15:
At first glance it looks to me like the behavior is still inconsistent, but not immediately clear.  Needs detailed explanation and/or test of current behavior to resolve (or exactly what to change in documentation).

jphickey, 2020-03-16:
The behavior of the `OS_rename` function while the file is open is a matter of the underlying OS and filesystem in use.  Some (maybe most?) OS's do allow this but some do not (VxWorks, maybe?).  

If anything, we might consider simplifying the code to remove the extra check, and run a quick test to document what happens on each of the supported platforms when attempting to rename a file that is open.

jphickey, 2020-03-16:
To clarify though - for the OSAL documentation we should just refer to the RTOS documentation, if available, for whether or not this is allowed.  Because even on the same OS the behavior can change for a variety of reasons (i.e. the file system type, its "mount" flags, read-only vs read-write, and others).

Even if we document what the MCP750 or Ubuntu Linux does, that might not hold true for other VxWorks platforms or other embedded Linux platforms that are still "posix" but employ different file systems.

skliper, 2020-03-16:
Copy.  How about at minimum:

Behavior of this API on an open file is not defined at the OSAL level due to dependencies on the underlying OS.  Some will allow the related operation and others will error.  Best practice is to limit use to files that are not open to maintain portability. 

Further documentation/testing is certainly allowed if someone has free time but I'd rather punt due to the complexities you note.

skliper, 2020-03-16:
Maybe improve the 2nd sentence - The underlying OS may or may not allow the related operation based on a variety of potential configurations.

jphickey, 2020-03-16:
The proposed wording sounds good to me...

PR #371 (2020-03-23): Fix #364 and others, Scrub user's guide
**Describe the contribution**
Document/comment changes only
Fix #364 - see issue for detailed changes
Fix #70 - since old doc was removed, and autodoc is correct

Fix #10 - replicated changes to old document in API header comments in https://github.com/nasa/osal/pull/371/commits/1aa87debb9bcb9424d6bf9ae87ca48f1f0aee199
Fix #225 - documented callback restrictions https://github.com/nasa/osal/pull/371/commits/4b19a1a91440525a073a4e6dcc6697867b7f4112
Fix #116 - document OS dependent behavior on open files https://github.com/nasa/osal/pull/371/commits/8f0780 [... cut]
commit 8f0780e9 (2020-03-23): Fix #116, Note undefined API behavior on open file

Comment/document only
closed 2020-03-27 by astrogeco