nasa/osal #120: vxworks osapi.c utility task doesn't exit

skliper, 2019-09-30:
The vxworks osapi.c has an optional task that is started in OS_API_Init():
{{{
/*---------------------------------------------------------------------------
 * Name:    UtilityTask
 * Purpose: If turned on, this task will print out the messages from
 *          the OS_printf buffer at a low priority. This will mean
 *          that the functions calling OS_printf will not block due to
 *          writing data to the UART
----------------------------------------------------------------------------*/
#ifdef OS_UTILITY_TASK_ON
void UtilityTask()
...
}}}

However, this utility task implementation has a while(TRUE) loop with no exit logic.  For a "load and forever run it" scenario this may work, but for any other scenario where an orderly shutdown is desired by the user this means the task is never killed.  One can get kernel crashes if the vxworks module is unloaded while that thread is still writing to output.  The forever loop thread also makes line coverage difficult.

We need a robust method to always end this thread and to ensure the parent thread doesn't return to OS control while this thread is still hanging out there.

skliper, 2019-09-30:
_Trac comment by abrown4 on 2015-11-25 17:11:19:_

See also #56, which adds an OS_ApplicationShutdown() and may run into the same issue when/if those changes are ported over to VxWorks.

skliper, 2019-10-22:
Note OS_ConsoleTask_Entry in posix/osapi.c is still a while(true) loop with no exit.

VxWorks task exits on semTake failure.  I think that means this task will exit if the semaphore is deleted, but I also don't see it being deleted.

jphickey, 2019-10-23:
There may be a simple fix here, by just checking the `ShutdownFlag` in the while loop.

PR #667 (2020-12-09): Fix #120, Exit console loop on shutdown
**Describe the contribution**
Fix #120 

Replaced condition on forever loops to end on shutdown. 

**Testing performed**
Built and ran unit tests for posix, built on vxworks (planning to build on RTEMs using the latest setup instructions...)

**Expected behavior changes**
- Loop will exit on shutdown

**System(s) tested on**
 - Hardware: cFS Dev server
 - OS: Ubuntu 18.04
 - Versions: Bundle main + this change

**Additional context**
What about also checking for a semaphore error on posix/rtems (or at least non-interrupt)?  Would make it consistent with  [... cut]
commit 9d523b67 (2020-12-09): Fix #120, Exit console loop on shutdown
commit d2d9882c (2020-12-09): Merge pull request #667 from skliper/fix120-forever-loop

Fix #120, Exit console loop on shutdown
PR #680 (2020-12-09): OSAL Integration Candidate: 2020-12-08
**Describe the contribution**

Fix #120, Exit console loop on shutdown
Fix #290, Remove obsolete printf tests
Fix #608, Add RTEMS 5.x support
Fix #631, add new functional tests for OS_chmod
Fix #648, scrub array references in shared layer
Fix #648, rename internal fields for consistency
Fix #648, sanitize array refs on impl layer
Fix #649, use full object ID in timecb list
Fix #650, OS_chmod uses read or write access.
Fix #664, change type of sync callback argument to osal_id_t

**Expected behavior changes**

PR #661 - Add support for RTEMS 5.1. The chang [... cut]
closed 2020-12-09 by astrogeco