nasa/osal #143: Support Insertion/Integration of Third Party/Bridge Libraries

skliper, 2019-09-30:
Missions/Projects may have the need/requirement to use specific implementations of functions provided by third party libraries i.e. ARINC 653 does not include a printf function.  

Third party libraries cannot legally be included within an official release of the OSAL.  The OSAL currently does not provide a generic means of integrating third party libraries into the OSAL.

See ticket #57 for more details

skliper, 2019-09-30:
_Trac comment by jphickey on 2015-12-02 12:23:35:_

I took a closer look .... turns out you can already do this very easily in the CMake build.  Both POSIX and RTEMS builds already use it.

* RTEMS has all cpu-specific routines in an `rtemscpu` library that needs to be explicitly specified on the link line
* POSIX puts dynamic loader functions in `dl`, thread functions in `pthread`.  Linux also puts real-time functions in a separate library called `rt` - none of these are in the standard C library, so they need to be explicitly specified as well

This is accomplished by setting the `OSAL_LINK_LIBS` variable to be a list of libraries to add to the final link.  It can be specified in the OSAL (per OS) and/or in the BSP (per board) or even in the external toolchain file.

Someone could put any 3rd party library in here; it is not limited to those supplied by the vendor/toolchain, so I believe this should work to support what is needed here.

Of course, one would have to use the CMake build system to get this feature.

If using the classic makefiles, then this is a matter of editing makefiles.  I don't think the CCB can provide a function for this, since classic makefiles are cloned and owned by the mission - maybe just note it in the README/docs where one would have to add 3rd party libraries in the provided sample makefile?

skliper, 2019-09-30:
_Trac comment by jphickey on 2015-12-02 12:38:09:_

The `OSAL_LINK_LIBS` list applies to //binary// libraries (i.e. built separately, prior to building CFE).

If someone has a //source// library that they would like to actually build and link in as part of a single CFE build, turns out we can do this too, via the PSP static module inclusion facility added by [cfs_psp:3]

All that needs to be done is to include the 3rd party library as a PSP module.  It can even be an external source tree outside of the CFE mission tree (build scripts support this too).

skliper, 2019-09-30:
_Trac comment by jphickey on 2015-12-02 12:39:19:_

To summarize - I don't think anything extra needs to be added here, there are plenty of hooks already there.  

This probably is just a documentation issue.

skliper, 2019-10-22:
Libraries are also supported.  No need for a 3rd method.

closed 2019-10-22 by skliper

jphickey, 2019-10-23:
> Libraries are also supported. No need for a 3rd method.

The CFE Library loading method is not useful here and does not support this use case, for systems like ARINC653, which need the 3rd party libraries for functions like printf().

The PSP module feature does support this use-case though, so as long as we keep that feature as part of the CMake build as it already is, I still think we are OK here.

skliper, 2019-10-23:
PSP module feature is not required to support 3rd party libraries.  It is not the only possible way to support 3rd party libraries for functions like printf().

jphickey, 2019-10-23:
PSP module is the only way to integrate a _SOURCE_ file that the CFE needs for itself to run (i.e. contains an implementation of some functions that PSP/CFE directly call, like printf), at least if you want that source file built and version-controlled as part of the CFE build.

For a pre-compiled _BINARY_ file, it can be added via OSAL_LINK_LIBS.

But the CFE runtime library loading is not sufficient for this use case, that was my point.  It is yet another example of why the PSP module feature is needed.

skliper, 2019-10-23:
Sorry, it's a non-starter to argue there is only one possible implementation.  Integration of a source file required by cFE can be accomplished by a multitude of mechanisms, not just your PSP module implementation.

jphickey, 2019-10-23:
Clarification: it is the way to address this issue that is currently implemented and available/working in the current master branch with the current CMake scripts.  Not that someone cannot implement yet another way...

There was some recent discussion of trying to reduce complexity and take this feature out, which is what I'm trying to make sure we do _NOT_ do, because many CFE use-cases do need this.

jphickey, 2019-10-23:
Not sure why this is so contentious?

The "solution" in the past was probably with the classic Makefile build and/or simply involved hacking the build scripts to add this.  One can always hack a makefile.... 

I think the point is, and the reason this issue was submitted in the first place, was the desire to have some way of doing it without requiring modification to the provided build scripts.

Again, the currently implemented "out of the box" way with the CMake scripts, to provide a library that CFE/PSP itself needs to run (like printf in this example):

- If it is a pre-compiled binary file, add it to OSAL_LINK_LIBS for that target.
- If its a source file/dir, add it in via the "targets.cmake" file in the TGTx_PSP_MODULELIST for that target

I don't know of any other way to add this without modifying the scripts themselves (which nobody will stop you if you want to do that).

We can rename the feature to something else, but this is precisely the use-case that this CMake build script hook is intended to serve.  It is not served by other methods that exist elsewhere.

skliper, 2019-10-23:
My concern is it's another feature lacking requirements and documentation that can easily be abused or used in a way that breaks concepts/scoping.

If there is source code needed by the core cFE/OSAL/PSP to work it shouldn't require changes within sample_defs to get it to work, you already select your platform.  If it's an OS thing (missing printf) handle it within the customized OSAL.  If it's a PSP thing, handle it within the customized PSP.  If it's anything outside of what's required to start up cFE, it should be in the runtime loading to support maintenance.

I'm not saying a user can't add to the core build if they so choose via including some random directory, but I certainly don't have a requirement to support it, document it, train people to use it, test it, etc.  All of this has associated cost.  If you provide an easy way to break the model (require adding "modules" in targets.cmake for your OSAL to work) then the stand alone OSAL is now broken.  People can and do use OSAL outside the context of cFE.

jphickey, 2019-10-23:
Where I disagree here is that, at a fundamental level, this is basically saying that in the process of deploying the system on any given HW, one will need to clone and own the PSP for that hardware, even if is almost entirely the identical an existing "out of the box" PSP.

Years ago, CCB discussions were had on this topic, and it was highly desirable to get away from this code-divergence model.

This might be fine for vxworks/rtems type targets, where everything is custom so you probably need a custom PSP anyway, but really doesn't work for the many embedded linux CubeSat variants out there.  These are all 90%+ the same, generally differing only in specific peripherals which are already abstracted to some degree via the Linux driver model.

Specific example: if I want to source my 1Hz CFE_TIME signal from GPS rather than emulating it from CPU timers like the default PSP does, I should be able to do this as part of my targets.cmake configuration as a local extension, just on the one CPU that has the GPS hardware.  I should not have to fork the entire PSP to do this.  

Insisting on a complete forking (or "cloning and owning") just to be able to tie into a platform-specific peripheral breaks compatibility with future framework-provided updates, and creates a maintenance headache for users going forward.

skliper, 2019-10-23:
This does not require an entire clone and own.  There are plenty of standard techniques available to use the provided code, and replace only what you need.  This can be done within PSP with no added "features" or requiring "module" support, that's the whole point.

jphickey, 2019-10-24:
Well, if so, these "standard techniques" are not documented nor are people aware of them, because the practice has been to clone and own the entire PSP just for minor additions/enhancements that are specific to one platform.  

I certainly am not sure what you are referring to -  I don't know how you'd add a local extension for to support something that is mission/hw/platform-specific, while otherwise using the "stock" PSP, without a hook in the build system to do so.

It would be good to document the expected approach for supporting custom extensions without cloning and owning, to get everyone on the same page, as this problem comes up frequently.

skliper, 2019-10-25:
I appreciate this exchange since it again reinforces the need for following the required project and design processes, along side a strong systems engineering and software quality assurance presence in the development activity.

I completely understand how you and your customers could categorize this feature as essential and a benefit.  The problem is from a context outside your direct support (you and your customers), this feature is not a requirement that has been costed, designed, documented, or implemented as required by our processes.  The biggest complaint I get from customers that are directly funding my oversight of this work is the build system is overly complex and undocumented.  There isn't a funding source that has requested documentation of standard techniques that can be used for developing PSPs, which I would happily do if resources are allocated.  It's outside the scope of my current resources, and taking on uncosted "requirements" is a recipe for failure.  If you'd like to contribute such documentation, or follow the required project and design process to get this feature officially supported you and your customers are encouraged to pursue it and fund it.

Unfortunately we are currently in a situation where a feature that has associated cost was not vetted through a project plan, and the required development process was not followed.  That is the issue I am funded to address.  You get what you pay for.

jphickey, 2019-10-25:
Not intending to keep dragging this issue out, I do understand your points too.  The problem I have is this:

-  "follow the required project and design process to get this feature officially supported" - the problem here is that, as best I can tell, there is no process for this.  GSFC basically dictates the CFE/OSAL requirements, and no input from community users is taken.

This feature is absolutely a requirement for the way GRC deploys/uses this software.  I maintain the CFE/CFS baseline for about a dozen different products - most are Linux-based.  This keeps it sustainable because they all share the same pc-linux PSP, the only difference is the set of extension modules on each platform.

GRC needed the feature as a solution for CFE deployment problems that they had.  Several GRC projects funded the development of it and contributed the code back to the community repo, which was ultimately accepted.

I understand that GSFC didn't/doesn't have a specific requirement for such a feature in their requirements.  But as I said, as far as I know there was not (and still is not) a forum for updating the requirements based on needs beyond those things specifically identified by GSFC only, even in instances where someone else in the community is willing to supply both the requirement and the implementation.

If the feature hasn't been implemented to the GSFC standards or adhere to the "required development process", I'm sure the GRC projects I support would be willing to kick in whatever extra effort is needed to meet those standards/process, because it is important to them.  But it seems there is no way to actually do this for CFE/CFS community users outside GSFC.

- "a feature that has associated cost" -- I still maintain this is a negative cost overall, as it allows de-scoping items from the framework, without alienating users that required those items.

skliper, 2019-10-25:
> * "follow the required project and design process to get this feature officially supported" - the problem here is that, as best I can tell, there is no process for this.

The governing process document is NPR 7150.2C.  The code is delivered as Class E, but shall be certifiable to class B by a project (meaning we need to provide all the required artifacts, traceability, design documentation, to support certification).

>  GSFC basically dictates the CFE/OSAL requirements, and no input from community users is taken.

The community most certainly can contribute and I encourage it!  Write and issue, propose requirements (or goals to enhance the product), request a feature and it will be evaluated.  They may get kicked up to the steering committee/architect review level, but that's a good thing by getting a system's engineering perspective.

> This feature is absolutely a requirement for the way GRC deploys/uses this software. I maintain the CFE/CFS baseline for about a dozen different products - most are Linux-based. This keeps it [...]

Excellent, then lets follow the process for evaluation to include the feature in a process compliant manner as part of the cFS Framework.  Not everything everyone proposes will be accepted as within the achievable/maintainable cFS Framework scope (we don't do everything GSFC projects want or even require either, we do what they fund).  For example, projects require a ground database but the cFS Framework doesn't have sufficient resources to maintain it for all the different systems out there so it remains a project cost.  It is a project requirement, not a cFS Framework requirement.

> GRC needed the feature as a solution for CFE deployment problems that they had. Several GRC projects funded the development of it and contributed the code back to the community repo, which was ultimately accepted.

Excellent, unfortunately a bottom up "feature" without the associated artifacts to be NPR 7150 compliant is not sufficient.

> I understand that GSFC didn't/doesn't have a specific requirement for such a feature in their requirements. But as I said, as far as I know there was not (and still is not) a forum for updating the [...]

Aren't you describing the issue review process we do on a weekly basis?  We entertain all sorts of feature requests, if funded or contributed they are considered, and with compliance to the process requirements those that get accepted go into the code base.  I don't recall ever cancelling/rejecting a pull request that was fully compliant.

> If the feature hasn't been implemented to the GSFC standards or adhere to the "required development process", I'm sure the GRC projects I support would be willing to kick in whatever extra effort is [...]

Certainly there is!  Start by stating the requirements/goals/use cases and present a design, document training guide(s) as needed, etc.  This would be a huge improvement over the current state!

> * "a feature that has associated cost" -- I still maintain this is a negative cost overall, as it allows de-scoping items from the framework, without alienating users that required those items.

I'm not aware of that negative cost being realized anywhere outside your scope of supported projects, since the design is not documented or communicated to the rest of the community sufficiently.  The feedback I get up to this point is adoption of cmake is a cost upper due to learning curve and lack of documentation.  Now don't get me wrong, I am a big fan of the build system in general due to months of digging through it but the projects I directly interact with don't have this sort of time or budget.  They need it to just work, and be simple to adapt to their use, but overall complexity of digging through it bottom up is virtually a non-starter without guides, design documentation, how-to's, etc.

jphickey, 2019-10-28:
> The governing process document is NPR 7150.2C. The code is delivered as Class E, but shall be certifiable to class B by a project (meaning we need to provide all the required artifacts, traceability, design documentation, to support certification).

OK- I think this is where the disconnect is occurring, possibly...

As far as I know, all of the process for NPR7150 Class E software was fulfilled here.  But perhaps the process wasn't as well-followed or documented in the early days of the CCB when this was added.

It was submitted and CCB discussed as PSP ticket 3 (babelfish) and discussed in CCB meetings staring at 2015-08-17.  The overall design was discussed at the CCB and ultimately pared down from the original proposed changeset, focusing on only modular build aspects and ultimately merged on 2015-09-08.   

During reviews at the time, the existence of this _build system_ feature was justified in that it supported the CFE requirement of portability to different hardware platforms.  Because it improved the applicability/adaptibilty to various Linux-based platforms it was seen as justified by requirements by the CCB at the time.

> Start by stating the requirements/goals/use cases and present a design, document training guide(s) as needed, etc. This would be a huge improvement over the current state!

> The feedback I get up to this point is adoption of cmake is a cost upper due to learning curve and lack of documentation

I am happy to go down this route, but in many ways (we need to recognize that this does not exist for the PSP or even the old/classic build system.  We had a wiki which contained a Cmake user guide, but the information was never brought over to github and I think most people didn't even bother to read it.

I would start by simply taking the existing wiki pages to get a reasonable starting point for a CMake howto.  I am willing to help with this task.  Rather than simply saying the documentation isn't sufficient, can we better identify the problem areas that people run into?

This conversation seems to have gone well outside the scope of this ticket (sorry for that!), but the initial reason I even commented here was to emphasize that this is a "need" that comes up from time to time both on the mailing list and the ticket system.  As people are trying to support various hardware platforms and configurations the need/applicability for something like this comes up.  If there isn't a requirement for it, there should be, because it is clearly useful to more than just the GRC projects I support.

But as it stands today, I'm still unclear how to get beyond this impasse.  There is no "community" requirements document for either the PSP nor the build system for which I could file a bug against.  (Only requirements for the CFE software itself are published).

So other than the documentation issue (which is much more widespread than just this one feature) what else is missing here to make this compliant and supported?

skliper, 2019-10-28:
Short answer:

> > Start by stating the requirements/goals/use cases and present a design, document training guide(s) as needed, etc. This would be a huge improvement over the current state!

Let's address at the CCB.