nasa/osal #165: UT assert library has uninitialized "UtTestDataBase" global variable

skliper, 2019-09-30:
The "UtTestDataBase" global, which is used as the primary node to the list of tests to run, is not initialized to a known state. 

This mostly works anyway due to the fact that most systems will zero out their BSS segment automatically while loading an exe, however some systems do not do this.

We should have an "init" function for UT assert that can be called during test setup, that can ensure that all globals are put to a known state.

skliper, 2019-10-22:
Or just initialize.

UtListHead_t UtTestDataBase = {.First=NULL, .Last=NULL, .NumberOfEntries=0};

jphickey, 2019-10-23:
> Or just initialize.
> 
> UtListHead_t UtTestDataBase = {.First=NULL, .Last=NULL, .NumberOfEntries=0};

This is already done implicitly.  AFAIK C99 says that uninitialized global data should be zeroed out by the implementation, which is effectively the same thing.

The problem is that on platforms like VxWorks, where execution is started by calling an entry point function from a shell, the test can be run more than once without reloading the executable.  

The first execution is generally OK, but subsequent runs fail due to the stale data in the global, which was never zeroed out because the executable wasn't reloaded.

skliper, 2020-12-03:
Is this OBE?  Looks like the list head is now part of UtAssert_Global, which gets initialized:

https://github.com/nasa/osal/blob/fa0cffaf4610d7bc11fc867aef325f5823c32fab/ut_assert/src/uttest.c#L140-L147

which is called from:

https://github.com/nasa/osal/blob/fa0cffaf4610d7bc11fc867aef325f5823c32fab/ut_assert/src/utbsp.c#L242-L245

If entry point on vxworks is OS_BSPMain, this clears OS_BSP_Global then calls OS_Application_Startup

Closing based on this, if further action necessary just reopen.  Many of the updates look to be from 1741655fad855cd68d4a83c08e2bd0fade18fad6 (fix #501)

closed 2020-12-03 by skliper

jphickey, 2020-12-04:
Yes - I can confirm this should no longer be an issue with the current implementation.