nasa/osal #166: stack_size parameter should be checked for reasonable value in OS_TaskCreate vxWorks implementation

skliper, 2019-09-30:
OS_TaskCreate calls vxWorks's taskSpawn() and passes the passed stack_size parameter to it. vxWorks does not appear to check this for a reasonable value. The OSAL should do that and then reject an unreasonable value (<4k or something) with an error code.

skliper, 2019-09-30:
_Imported from trac issue 143.  Created by sseeger on 2016-01-06T09:47:44, last modified: 2019-08-14T14:11:46_

skliper, 2019-09-30:
_Trac comment by glimes on 2016-01-14 12:10:36:_

See also #167 for what happens when we do not check.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-03 13:34:40:_

Moved open 4.2.2 tickets to 4.3.1

skliper, 2019-10-22:
Arbitrary limit not required.  Suggest keeping code as is (will fail if there isn't space available).

jphickey, 2019-10-23:
Concur/agree that this is OK as-is.  The OSAL should do what it was instructed to do.  Calls to OS_TaskCreate should just specify the stack size that the task needs, and not try to fix mistakes of the caller.

skliper, 2019-10-23:
Closing as wontfix.

closed 2019-10-23 by skliper