nasa/osal #176: utlist can use malloc(0) and memcpy of 0 size data

skliper, 2019-09-30:
It's ok for memcpy to use 0 byte data, but the behavior of malloc with 0-byte data is inconsistent across platforms. This is not a required fix, however some static code analyzers might complain. 

(The size of the malloc is dependent on data passed into it.)

[changeset:93b15ea] ready for CCB review

skliper, 2019-09-30:
_Imported from trac issue 153.  Created by sseeger on 2016-01-10T23:07:30, last modified: 2019-08-14T14:11:46_

skliper, 2019-10-22:
Looks like this is specific to ut_assert, in utlist.c.  Unit test specific issue.

Not sure what resolution should be - skip malloc and memcpy if DataSize == 0?  Or just print an warning?

skliper, 2020-12-03:
No malloc/memcpy 0 possibility seen in src/utlist.c anymore.  It now does just one malloc including ListAllocator and NodeAllocator unions at minimum (along with the data).  Looks like it was fixed in 9a9171c5073c8b27975e8d16218556335474305b, fix #550.

skliper, 2020-12-03:
Note - spot checked for other malloc(0) potential and none were found.  UT_SetDataBuffer has a malloc(BufferSize) but it's checked for nonzero before calling.

closed 2020-12-03 by skliper

jphickey, 2020-12-04:
Yes - I can confirm this should no longer be an issue with the current implementation.