nasa/osal #180: Clear BSS on App Restart

skliper, 2019-09-30:
Under normal conditions, when a task is started, the process of loading
the task will clear the BSS segment (used to store global-scope variables
that are not explicitly initialized).

However, if a task is "restarted" without "reloading" it may find that
its BSS segment still retains the content from the prior run.

We discussed this at the CCB meeting on 2016-Jan-19, and the moderate
concensus was that it might be most useful for the BSP to explicitly clear
the BSS of tasks before they are started. According to my notes, Joe was
going to take a look at this.

skliper, 2020-12-03:
@jphickey - Any updates on this in the context of Caelum?  If we want to defer I'll probably need a justification.

jphickey, 2020-12-04:
There are two big issues here, namely that:

1. It is supposed to be done by the OS/loader implementation (in fact c99 requires it AFAIK)
2. Because of (1) There is no standards-based method for an application to get the .bss address+size and clear on its own.

To ensure we don't have any issues, an application should do two things here, as part of the recommended pattern:

- Put all its variables into a single global structure named according to the app (i.e. CFE_ES_AppData, SAMPLE_AppData, etc).
- `memset()` that structure to zero as one of the first actions in the entry point function.

We should already have that global structure implemented in all the framework apps.  Not only does it establish a single reference object that represents all global data, it gives a logical namespace separation as the global is named with the app name and therefore should be unique.  Members of that struct can then have the same name as other apps and they will not conflict.

While the structure exists in all framework apps we should probably add the `memset()` to make it a complete example.

skliper, 2020-12-04:
Copy, I was struggling to figure out how this would be done in OSAL based on 2.  Suggested pattern makes more sense to me.  Moving to CFE to confirm the pattern is applied everywhere (forward action to review lab apps also).