nasa/osal #182: Add API to read tick counter from TimeBase object

skliper, 2019-09-30:
The "TimeBase" OSAL API (already merged) should include the ability to read the internal counter value.

This can cover the same use-cases that the old trac #32 was targeting, without the pitfalls that were brought up with that solution.

skliper, 2019-09-30:
_Trac comment by jphickey on 2016-02-19 10:05:20:_

Pushed commit [changeset:6bfd48f] to address this.

**DEPENDENCY NOTE**: Only the timebase API was merged into development, not the implementation.

This depends on trac #28 (ng OSAL) which is where the timebase is actually implemented.  Therefore this is a descendent of #28 and cannot be merged before that one.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-04-05 17:45:37:_

Is this ready for code review?  Does it need a rebase?  Or was it OBE with the #28 merge?

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-03 13:47:08:_

Moved 4.3.0 tickets that won't make release to 4.4.0

skliper, 2020-01-02:
Is this still needed?  I didn't see an API to read a tick w/ quick inspection.

jphickey, 2020-01-02:
This is already implemented as `OS_TimeBaseGetFreeRun()`

The output of this function is a simple 32-bit unsigned counter, and hence can be easily used for relative interval measurements (take two samples, subtract old from new).

This ticket can be closed

closed 2020-01-02 by jphickey

skliper, 2020-01-02:
Marked as a duplicate of #28 for historical/VDD reasons.