nasa/osal #194: Missing Documentation on HowTo Build/Run Unit and Functional Tests

skliper, 2019-09-30:
At a minimum a simple readme file should be included under the functional, black box, and white box unit test directories to provide users with some guidance on how to build and run each set of tests.

skliper, 2019-09-30:
_Trac comment by abrown4 on 2016-08-23 18:20:43:_

FYI, Added the helpful script used for testing #189 on pc-linux.  Run twice to test both the pc-linux mqueue and socket implementations.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-03 13:34:40:_

Moved open 4.2.2 tickets to 4.3.1

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-26 15:13:16:_

Milestone renamed

CDKnightNASA, 2020-03-31:
While I'd love to write this documentation, I actually don't know how to build OSAL standalone with cmake. Joe?

jphickey, 2021-04-02:
There is a "Quick Start" section about how to do this in the README.  it is somewhat buried toward the bottom, after the version history, so its possible that most users don't see it there.  Basically it says:

```
mkdir build_osal_test
cd build_osal_test
cmake -DENABLE_UNIT_TESTS=true -DOSAL_SYSTEM_BSPTYPE=generic-linux -DOSAL_CONFIG_DEBUG_PERMISSIVE_MODE=TRUE ..
make
make test
```

I just confirmed these commands do in fact still work.  I can build OSAL standalone from CFE, using the above command, and "make test" runs the tests.  

Other than the issue of visibility - I do think this "quick start" section should be more prominent - is there more info that we need to put here?  The commands seem to work for me, and seems clear enough.

closed 2021-04-02 by skliper