nasa/osal #195: VxWorks OSAL uses potentially unsafe "strcpy" operations

skliper, 2019-09-30:
In at several locations the VxWorks OSAL is using `strcat()` and `strcpy()` functions to copy strings between string buffers that are sized using different macros.

At least one of the sizes in play, the `OS_MAX_PATH_LEN` comes from the user-configurable "osconfig.h" file.  Other sizes, such as `OS_FS_PHYS_NAME_LEN` are specified in the local headers and are **not** user-configurable.

In some functions, such as `OS_mkfs` (but not limited to this), a local buffer of size `OS_MAX_PATH_LEN` is copied into a global buffer of size `OS_FS_PHYS_NAME_LEN`.  

However, because the `OS_MAX_PATH_LEN` is configurable via the osconfig.h file, it is not guaranteed that `OS_MAX_PATH_LEN` is less than or equal to `OS_FS_PHYS_NAME_LEN`.

skliper, 2019-09-30:
_Trac comment by jphickey on 2019-03-26 08:52:42:_

This can be added to the pool of "stuff that gets fixed by migrating to NG architecture".  After #231 gets merged this should be closed.

skliper, 2019-10-22:
Just needs close review to confirm all cases were addressed.

skliper, 2020-01-02:
Only remaining concern:

os/shared/osapi-printf.c:        strcpy(console->device_name, OS_PRINTF_CONSOLE_NAME);

Could be defined in osconfig.h, but no check on size seen (quick inspection).

skliper, 2020-04-20:
Want me to reassign this one? @jphickey

jphickey, 2020-04-20:
No, I'll just change that last case to an strncpy instead, as part of the other refactoring I am doing for #285.  Any change made separately will likely cause merge woes.

Note that the symbol OS_PRINTF_CONSOLE_NAME would be defined in osconfig.h, which the same file as OS_MAX_API_NAME (the actual length limit) is defined.  But it doesn't hurt to use strncpy instead of strcpy here just in case the user misconfigures, and its a trivial change in the code.

skliper, 2020-06-05:
Closed in #443

PR #443 (2020-05-04): Fix #285, Refactor OSAL to avoid inclusion of C files
**Describe the contribution**
Use separate source files and CMake-based source selection
based on feature configuration, rather than using the C
preprocessor for including/excluding different OSAL function
groups.

Refactor all implementation units to provide a separate header
file for each functional group/subsystem.  Remove "static"
declaration on internal helper functions so they can be invoked
from unit test.

Fixes #285 
Also Fix #214 and fix #195 (trivial fixes rolled in as part of refactoring)

**Testing performed**
Build and execute [... cut]
closed 2020-06-05 by skliper