nasa/osal #197: Black Box Unit Tests Do Not Ensure Resources Are Cleaned Up Between Tests

skliper, 2019-09-30:
Unit tests should utilize a SetUp and TearDown approach to ensure each test runs in an expected and known configuration.

The TearDown function needs to ensure that all test objects and resources have been deallocated and cleanup.

skliper, 2020-12-03:
ut_assert now has UtTest_AddStartup and UtTest_AddTeardown to support setup and teardown.  Action to close this issue is to scrub unit tests to confirm cleaning up.

jphickey, 2021-04-02:
One potential issue here is that we don't have the equivalent teardown function for e.g. `OS_API_Init()`.  Once we set up OSAL it is expected to remain set up until a reboot.  While all user-created objects in OSAL can be deleted, the OSAL itself and its internal objects cannot be cleaned up via the current API.

If the intent is to allow multiple tests to be run on system like VxWorks, without a reboot in between, then we'll likely need to implement #871 to allow this to happen.

Otherwise all user objects can be easily cleaned up by making sure every test program has a Teardown function that invokes `OS_DeleteAllObjects()`  (this shouldn't be a big deal to implement).

PR #948 (2021-04-11): Fix #944, add OS_API_Teardown
**Describe the contribution**
This cleans up all OSAL resources as best as possible, ideally leaving the system in a state where OS_API_Init() may be invoked again.

Fixes #944
Fixes #197 

**Testing performed**
Build and run all unit tests, confirm correct behavior
Test in conjunction with fix for #197, confirm test can be run multiple times on VxWorks without reboot

**Expected behavior changes**
Adds OS_API_Teardown API to complement OS_API_Init

**System(s) tested on**
Ubuntu 20.04 (native)
MCP750 vxworks 6.9

**Contributor Info - All information REQUIRED for  [... cut]
commit ef34cf49 (2021-04-11): Fix #197, add teardown at end of tests

Invoke OS_API_Teardown to reset state for next test
PR #956 (2021-04-12): osal Integration candidate: 2021-04-13
## Description

### PR #941 

Fix #871, allow OSAL re-initialization
Fix #691, Combine initialized and shutdown flags into one

Replaces the separate "Initialized" and "Shutdown" flags with a single state flag.  Creates a global single source of truth for the OSAL state. This enables users to run tests and `OS_API_Init()` multiple times without a reboot in the middle to reset the state.
- Multiple invocations of `OS_API_Init()` are allowed - subsequent calls can be ignored
- Deleting of any internal objects that did get created if `OS_API_Init()` fails (t [... cut]
closed 2021-04-12 by astrogeco