nasa/osal #213: OS_ModuleLoad should copy the module before loading.

skliper, 2019-09-30:
Many OS's will allow you to modify/overwrite a shared library and any running executables that have loaded that library will be immediately affected (likely causing a crash.) Clearly missions will want to upload updates to modules and not have them affect the running cFE.

A simple solution is to copy the module library file to a temporary file. Yes this will increase the storage space required, so a #define switch to enable this behavior should be used to ensure folks know this is going to happen (or they risk crashing the cFE if they upload a module file directly over the existing module file.)

(Note that file locking mechanisms are unreliable in many OS's.)

skliper, 2019-10-22:
Not clear that this is applicable.  Modify/overwrite of a current library isn't supported due to all the possible challenges/issues and complexity involved in doing it correctly.

skliper, 2019-10-22:
Recommendation is a reset, and replace the library in the startup file.

jphickey, 2019-10-23:
> Not clear that this is applicable. Modify/overwrite of a current library isn't supported due to all the possible challenges/issues and complexity involved in doing it correctly.

Concur that it isn't supported, but the issue is that it happens "accidentally" and catches some users off-guard as they were not expecting this behavior.  Note that `OS_ModuleLoad` applies to both libraries and applications, the latter of which is supported to unload/reload at runtime.  

The issue here is that operating systems that employ virtual memory/address translation will not actually load the content of the .so file in storage (an app or lib) into RAM, they just map pages to the disk file itself, and caches pages in RAM as they are used.

As a result, if the app/lib file in the storage is updated "in-place" by some process external to CFE (like FTP or SCP), the changes are immediately visible to the running instance, without any explicit reload.  This is obviously not desired or reliable, and likely to crash the system in many cases, particularly while the FTP/SCP operation is in progress.

It is fairly easy to mitigate this externally as well.  Just use a proper update tool that does not update the target file in-place, but rather makes a new temporary copy and then renames it to the old name once the complete file is transferred.

On linux targets, the "scp" tool is known to do this the wrong way (in-place) where the "rsync" tool does it the correct way, at least by default.

My recommendation is to just document this somewhere that any external application that does in-flight software updates should always write to a NEW temporary file and then RENAME it to the old name, rather than updating a file in place.  But this should be standard practice anyway, because you don't want a half-updated file should the transfer fail.

skliper, 2019-10-23:
CCB 20191023 - Consequences of coping modules in the module load traded against operational solutions, change not worth the additional complexity.

closed 2019-10-23 by skliper