nasa/osal #219: semaphore APIs do not allow for desired signals to be handled

skliper, 2019-09-30:
From the original reporter:
"The call to OS_InterruptSafeLock() blocks all signals in order to force
mutual exclusion with signal handlers. However, it is often the case that
the semaphore we are waiting on is given inside a signal handler, which
now is never executed because the signals are blocked. This results in
deadlock."

skliper, 2019-09-30:
_Imported from trac issue 196.  Created by tbrain on 2018-04-03T14:42:29, last modified: 2019-08-14T14:11:46_

skliper, 2019-09-30:
_Trac comment by jwilmot on 2018-07-24 14:19:26:_

Included as part of #28.

skliper, 2019-09-30:
_Trac comment by jphickey on 2019-03-05 11:59:06:_

There is one particular phrase of this description which I think deserves a discussion of its own:

> However, it is often the case that the semaphore we are waiting on is given inside a signal handler

Admittedly, this has been rather poorly documented in OSAL, but most abstraction functions are generally **not** ISR or signal safe.  As a general rule, the only API that is (probably?) async-signal safe on all operating systems is the counting semaphore give function, not the binary one.

That being said, I am curious about how the observation came about, as OSAL does not directly expose any API that deal directly with signal handlers.   The only exception is the original timer implementation which did indirectly (perhaps mistakenly?) expose a signal handler context via the timer callback function.  

If the issue is related to giving a semaphore inside a timer callback, then this issue is solved by doing a better timer implementation that doesn't give callbacks directly from a signal handler.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-03-05 13:19:15:_

See also discussion on #207

skliper, 2019-09-30:
_Trac comment by jphickey on 2019-03-26 08:55:42:_

Adding this to the group of bugs that are fixed by using the "NG" architecture

skliper, 2019-09-30:
_Trac comment by tbrain on 2019-04-23 16:20:46:_

Replying to [comment:3 jphickey]:

> That being said, I am curious about how the observation came about, as OSAL does not directly expose any API that deal directly with signal handlers.   The only exception is the original timer [...]

I'm not sure what else to say except that I added an API to associate allowed signals to specified semaphores 

{{{
int32 OS_BinSemAddSignalMask    (uint32 sem_id, int sigid);
}}}

Does that answer your question?

skliper, 2019-09-30:
_Trac comment by jphickey on 2019-04-23 16:50:35:_

Replying to [comment:7 tbrain]:
> I'm not sure what else to say except that I added an API to associate allowed signals to specified semaphores 
> 
> {{{
> int32 OS_BinSemAddSignalMask    (uint32 sem_id, int sigid);
> }}}
> 
> Does that answer your question?

Not quite -- I still don't see the root of the issue here, because the concept of "signals" in general is supposed to be abstracted away by OSAL and/or the PSP.  There should be no application code that relies directly on signals, since it is not entirely portable.

In a nutshell, this whole issue strikes me as a //symptom// of some other underlying problem.  If OSAL were doing its job, this wouldn't be needed in the first place.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-26 15:13:05:_

Milestone renamed

jphickey, 2020-10-30:
Looking through old issues I see this one assigned to me and still open.... As far as I can tell this was a remnant of the old/last-gen OSAL 4.2.x and this is not applicable to the current OSAL 5.1.x.

Note that in the current OSAL, all async signals should be blocked in OSAL-created tasks.  The only task with signals enabled is the original (root) task to handle such events as CTRL+C and this is all done by the PSP.  Signals shouldn't be used or relied upon in any CFE app intended to be portable.

Will close this one as no longer applicable -- please re-open or re-submit a new issue against the current OSAL if there is still a problem.

closed 2020-10-30 by jphickey