nasa/osal #229: mqueue test program

skliper, 2019-09-30:
OSAL should include a simple mqueue test program to validate that the user has the correct settings and permissions to create/open/close/delete mqueues. Often users stumble on mqueue configuration and it is more difficult to diagnose when it's wrapped in the entirety of OSAL/cFS.

skliper, 2019-09-30:
_Imported from trac issue 206.  Created by cdknight on 2018-11-16T19:04:20, last modified: 2019-08-14T14:11:46_

jphickey, 2019-10-18:
Rather than making a whole new test program, there is already a "queue-timeout-test" test program that does much of this.  

Right now this test program only focuses on calling OS_QueueCreate() followed by OS_QueueGet() and getting the timeout response.  But it shouldn't be too hard to just add another test case to this, to actually call OS_QueuePut() to send some test data into the queue and validate it on the other side.

Furthermore, the ut_oscore_queue_test does a much more extensive test of the OS_QueuePut/OS_QueueGet functions, but it is a bit convoluted and doesn't use the same framework.

skliper, 2019-10-18:
Preference is to add test or add to existing test (queue-timeout-test) using the utassert framework.  Either approach is fine (if you want to keep queue-timeout-test specific to timeouts and add a test for general queue behavior, or if you want to generalize the existing test to cover more functionality).  

Expending resources on the "unit-tests" that use the alternate framework is discouraged.  Eventually the "scenarios" covered by these tests should all be covered by the utassert tests.

PR #695 (2021-01-25): Fix #229, mqueue functional test
**Describe the contribution**
Fixes #229 
Generalized queue timeout test to also test message queue functionality 

**Testing performed**
Build and run the unit tests 

**Expected behavior changes**
Adds 2 seconds to the unit tests 

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Alex Campbell GSFC
commit bddede21 (2021-01-25): Merge pull request #695 from zanzaben/fix229_mqueue_functional_test

Fix #229, mqueue functional test
commit e9638478 (2021-01-25): Fix #229, add mqueue functional test
commit f725ee41 (2021-01-25): Fix #229, mqueue test uses sequence of numbers
closed 2021-01-27 by astrogeco