nasa/osal #232: PTHREAD_STACK_MIN typo in POSIX/POSIX-NG OSAL

skliper, 2019-09-30:
Both the POSIX and POSIX-NG OSALs contain the following in `osapi.c`:
{{{#!c
#ifndef PTHREAD_STACK_MIN
   #define PTHREAD_STACK_MIN 8092
#endif
}}}

I believe this is supposed to be 8192 or 8*1024.

skliper, 2019-09-30:
_Imported from trac issue 209.  Created by tflemin1 on 2019-03-28T14:41:32, last modified: 2019-08-14T14:11:46_

davearch, 2020-03-20:
Hello, can I fix this issue? Seems simple enough

skliper, 2020-03-23:
@davearch Absolutely!  Note we need one of the following to cover "external contributions" (there's also a link on the pull request template, submission instructions on the form):
https://github.com/nasa/osal/blob/master/doc/GSC_18370_Corp_CLA_form_1219.pdf
https://github.com/nasa/osal/blob/master/doc/GSC_18370_Ind_CLA_form_1219.pdf

davearch, 2020-03-23:
@skliper Thank you so much! So to be clear, I should fill out the individual contributor form and mail it in before I submit a pull request?

davearch, 2020-03-24:
Oh I just saw I can email it I will do that

skliper, 2020-03-24:
@davearch email is preferred.  Order isn't all that critical, we just mark a pull request as needs CLA until we can confirm it has been received.  Individual is the right form as long as you are doing it on your time.  Corporate form is for contributions done as part of your job.

PR #402 (2020-04-15): Fix #232, pthread stack min typo
**Describe the contribution**
Fixes #232 PTHREAD_STACK_MIN typo. I could have changed the number to 8192 but I thought (8*1024) was more true to where the number comes from. Let me know if this is not inline with NASA code style preferences.

**Testing performed**
Steps taken to test the contribution:
1. Build steps: built with cmake:
`cmake -DOSAL_SYSTEM_OSTYPE=posix -DOSAL_SYSTEM_BSPTYPE=pc-linux -DENABLE_UNIT_TESTS=TRUE -DOSAL_INCLUDEDIR=/home/david/projects/config -DCFE_SYSTEM_PSPNAME=whyohwhy /home/david/projects/osal`
1. Ran all tests and unit tests.
Note [... cut]
commit 2f81e64e (2020-04-15): fixes #232 PTHREAD_STACK_MIN typo
commit cf4ef01f (2020-04-15): Merge pull request #402 from davearch/fix-232-PTHREAD_STACK_MIN-typo

Fix #232, pthread stack min typo
PR #417 (2020-04-20): Integration Candidate: 2020-04-15
**Describe the contribution**
Fixes #232, #397, #312, #261, #362, and #363

See Bundle IC at https://github.com/nasa/cFS/pull/66

**Testing performed**
See PRs
Bundle CI - https://travis-ci.com/github/nasa/cFS/builds/160607248

**Expected behavior changes**
- PR #402 - Fixed typo, no behavior changes
- PR #405 - Code cleanup, no behavior changes
- PR #404: 
  - Use the target_include_directories and target_compile_definitions functions from CMake to manage the build flags per target. 
  - Build implementation components using a separate CMakeLists.txt file rat [... cut]
closed 2020-04-20 by astrogeco