nasa/osal #265: Remove references to OSAL_OPAQUE_OBJECT_IDS and OSAL_ABSTRACT_FILESYS_TYPES

skliper, 2019-10-16:
**Is your feature request related to a problem? Please describe.**
References to removed #defines remain in the unit test code.

https://github.com/nasa/osal/blob/ad88aa16625d998efd7473fea6e347ccfe097e19/src/unit-test-coverage/vxworks-ng/CMakeLists.txt#L5-L6

**Describe the solution you'd like**
Remove references

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

waheedejaz, 2020-01-15:
Can I grab this issue?

jphickey, 2020-01-15:
Hold off on this one... in the fix for issue #312 (implemented but not yet in a pull request) refactored some of the same file where this appears so it was removed in the process.

jphickey, 2020-04-20:
Confirmed this is already fixed and merged in pull request #404, commit 4904591.  There are no occurrences of either `OSAL_OPAQUE_OBJECT_IDS` or `OSAL_ABSTRACT_FILESYS_TYPES` in the current master branch.

closed 2020-04-20 by jphickey