nasa/osal #283: cpp check error

avan989, 2019-11-05:
**Describe the bug**
Cppcheck returns warning:

[osal/src/os/posix/osapi.c:1832]: (error) Shifting signed 32-bit value by 31 bits is undefined behaviour
[osal/src/unit-test-coverage/shared/src/coveragetest-time.c:175]: (error) Shifting signed 32-bit value by 31 bits is undefined behaviour
[osal/src/unit-test-coverage/shared/src/coveragetest-timebase.c:111]: (error) Shifting signed 32-bit value by 31 bits is undefined behaviour
[osal/src/unit-test-coverage/ut-stubs/src/libc-stdio-stubs.c:145]: (error) Uninitialized variable: actual
[osal/src/unit-test-coverage/ut-stubs/src/libc-stdio-stubs.c:167]: (error) Uninitialized variable: actual
[osal/src/unit-tests/oscore-test/ut_oscore_binsem_test.c:165]: (error) syntax error
[osal/src/unit-tests/oscore-test/ut_oscore_countsem_test.c:166]: (error) syntax error
[osal/src/unit-tests/oscore-test/ut_oscore_exception_test.c:173]: (error) syntax error
[osal/src/unit-tests/oscore-test/ut_oscore_interrupt_test.c:193]: (error) syntax error
[osal/src/unit-tests/oscore-test/ut_oscore_misc_test.c:173]: (error) syntax error
[osal/src/unit-tests/oscore-test/ut_oscore_mutex_test.c:166]: (error) syntax error
[osal/src/unit-tests/oscore-test/ut_oscore_queue_test.c:169]: (error) syntax error
[osal/src/unit-tests/oscore-test/ut_oscore_task_test.c:208]: (error) syntax error
[osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c:208]: (error) syntax error
[osal/src/unit-tests/osfile-test/ut_osfile_fileio_test.c:304]: (error) syntax error
[osal/src/unit-tests/osfilesys-test/ut_osfilesys_diskio_test.c:220]: (error) syntax error
[osal/src/unit-tests/osloader-test/ut_osloader_module_test.c:167]: (error) syntax error
[osal/src/unit-tests/osloader-test/ut_osloader_symtable_test.c:158]: (error) syntax error
[osal/src/unit-tests/osnetwork-test/ut_osnetwork_misc_test.c:181]: (error) syntax error
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:74]: (error) sprintf format string requires 1 parameter but only 0 are given.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:81]: (error) snprintf format string requires 1 parameter but only 0 are given.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:394]: (error) va_list 'Ptr' used before va_start() was called.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:417]: (error) va_list 'Ptr' used before va_start() was called.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:443]: (error) va_list 'Ptr' used before va_start() was called.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:469]: (error) va_list 'Ptr' used before va_start() was called.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:494]: (error) va_list 'Ptr' used before va_start() was called.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:508]: (error) va_list 'varg' used before va_start() was called.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:522]: (error) va_list 'varg' used before va_start() was called.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:537]: (error) va_list 'varg' used before va_start() was called.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:552]: (error) va_list 'ptr' used before va_start() was called.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:565]: (error) va_list 'ap' used before va_start() was called.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:584]: (error) va_list 'Ptr' used before va_start() was called.
[osal/src/unit-tests/osprintf-test/ut_osprintf_offset.c:607]: (error) va_list 'Ptr' used before va_start() was called.
[osal/src/unit-tests/ostimer-test/ut_ostimer_timerio_test.c:206]: (error) syntax error

**To Reproduce**
1. cppcheck --force .
2. See warning

**System observed on:**
Hardware
1. 18.04
2. Cppcheck 1.82, cFE 6.7.1, osal 5.0.1

**Reporter Info**
Anh Van, NASA Goddard

jphickey, 2019-11-05:
I will look into these:

[osal/src/os/posix/osapi.c:1832]: (error) Shifting signed 32-bit value by 31 bits is undefined behaviour
[osal/src/unit-test-coverage/shared/src/coveragetest-time.c:175]: (error) Shifting signed 32-bit value by 31 bits is undefined behaviour
[osal/src/unit-test-coverage/shared/src/coveragetest-timebase.c:111]: (error) Shifting signed 32-bit value by 31 bits is undefined behaviour
[osal/src/unit-test-coverage/ut-stubs/src/libc-stdio-stubs.c:145]: (error) Uninitialized variable: actual
[osal/src/unit-test-coverage/ut-stubs/src/libc-stdio-stubs.c:167]: (error) Uninitialized variable: actual

For the rest (i.e. everything under "unit-test") all this code needs a general rework.  I believe there is a separate ticket about that.  I wouldn't recommend fixing these warnings until the rework is done (or better, fix them as part of the rework).

skliper, 2019-11-06:
I would like these all fixed now since they don't look to hard to solve so we can start enforcing no cppcheck errors at the bundle level in CI since major rework will likely take a while.

skliper, 2019-11-06:
Feel free to split the ticket if it facilitates resolution.

closed 2019-11-12 by avan989

skliper, 2020-01-08:
Split