nasa/osal #285: Replace #include of c files with build system source selection

skliper, 2019-11-08:
**Is your feature request related to a problem? Please describe.**
#include of a c file is not acceptable flight coding style for some organizations and complicates formal verification processes

**Describe the solution you'd like**
Source selection should be done via build system

**Describe alternatives you've considered**
None.

**Additional context**
None.

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2019-12-22:
Moved from #327

osal/src/os/posix/osfileapi.c
```
*/
--
81 | #include "../portable/os-impl-posix-io.c"
  | The #include pre-processor directive should only be used to include header files.

*/
88 | #include "../portable/os-impl-posix-files.c"
  | The #include pre-processor directive should only be used to include header files.
89

*/
--
95 | #include "../portable/os-impl-posix-dirs.c"
  | The #include pre-processor directive should only be used to include header files.
```
posix/osapi.c:
```
/* use the portable version of OS_ConsoleWrite_Impl() */
--
2411 | #include "../portable/os-impl-console-directwrite.c"
  | The #include pre-processor directive should only be used to include header files.
```
posix/osloader.c
```
*/
--
32 | #include "../portable/os-impl-posix-dl.c"
  | The #include pre-processor directive should only be used to include header files.
```

posix/osnetwork.c
```
*/
--
58 | #include "../portable/os-impl-bsd-sockets.c"
  | The #include pre-processor directive should only be used to include header files.
```

posix/osselect.c
```
/* Leverage the portable UNIX-style Select API for POSIX */
--
35 | #include "../portable/os-impl-bsd-select.c"
  | The #include pre-processor directive should only be used to include header files.
```
posix/ostimer.c
```
/* POSIX implements clock_gettime and clock_settime that can be used */
--
566 | #include "../portable/os-impl-posix-gettime.c"
  | The #include pre-processor directive should only be used to include header files.
```

jphickey, 2020-03-10:
A "work in progress" concept is pushed to my osal fork, for review and discussion:

See: https://github.com/jphickey/osal/tree/jph-wip-fix-285-includes

skliper, 2020-04-15:
High priority change, submit the pull request on top of other pending merges if you have to.

commit 908709a3 (2020-04-29): Fix #285, Refactor OSAL to avoid inclusion of C files

Use separate source files and CMake-based source selection
based on feature configuration, rather than using the C
preprocessor for including/excluding different OSAL function
groups.

Refactor all implementation units to provide a separate header
file for each functional group/subsystem.  Remove "static"
declaration on internal helper functions so they can be invoked
from unit test.
commit 14312324 (2020-04-29): Fix #285, Refactor OSAL to avoid inclusion of C files

Use separate source files and CMake-based source selection
based on feature configuration, rather than using the C
preprocessor for including/excluding different OSAL function
groups.

Refactor all implementation units to provide a separate header
file for each functional group/subsystem.  Remove "static"
declaration on internal helper functions so they can be invoked
from unit test.
commit 7a7e6d1f (2020-04-29): Merge pull request #427 from jphickey/fix-285-code-selection
Fix #285, Refactor OSAL code selection
commit edab36d7 (2020-04-30): Merge pull request #427 from jphickey/fix-285-code-selection

Fix #285, Refactor OSAL to avoid inclusion of C files
Manually merged to address merge conflicts
closed 2020-05-05 by astrogeco