nasa/osal #287: cppcheck coverage check warning

avan989, 2019-11-12:
**Describe the bug**
reference issue #283 

[osal/src/os/posix/osapi.c:1832]: (error) Shifting signed 32-bit value by 31 bits is undefined behaviour
[osal/src/unit-test-coverage/shared/src/coveragetest-time.c:175]: (error) Shifting signed 32-bit value by 31 bits is undefined behaviour
[osal/src/unit-test-coverage/shared/src/coveragetest-timebase.c:111]: (error) Shifting signed 32-bit value by 31 bits is undefined behaviour
[osal/src/unit-test-coverage/ut-stubs/src/libc-stdio-stubs.c:145]: (error) Uninitialized variable: actual
[osal/src/unit-test-coverage/ut-stubs/src/libc-stdio-stubs.c:167]: (error) Uninitialized variable: actual

**Reporter Info**
Anh Van, NASA Goddard

skliper, 2019-11-12:
@jphickey Anh split the ticket, let me know if you foresee any challenges in getting this resolved quickly.

skliper, 2019-11-16:
@jphickey will you be able to address this one soon or should I reassign?  It's a hold on us enforcing no cppcheck errors in CI.

commit 0f10e0ea (2019-11-19): FASTTRACK: Merge #291

Fixes #287: cppcheck warnings
commit cc9e567d (2019-11-19): Fix #287: cppcheck warnings

Replace use of (1<<31) and (INT_MAX) with (UINT32_MAX/2) which should
squelch the cppcheck warnings about the left shift as well as being
more correct (the values in question are uint32's, not int's).

Also address two false positives about a return value being unassigned,
by simply initializing the value to be 0.
PR #292 (2019-11-30): Integration Candidate 20191115
**Describe the contribution**
Fixes #280, fixes #286, fixes #287

**Testing performed**
Steps taken to test the contribution:
1. Checked out bundle with ic-20191115 branches 
1. make ENABLE_UNIT_TESTS=TRUE SIMULATION=native prep
1. make
1. make install
1. make test
Built without warnings, all tests passed except osal_timer_UT (nominal result on linux)
executed cfe, successful startup with no warnings

Confirmed passed CI (with cppcheck enforced!)

**Expected behavior changes**
No functional changes, all doxygen/cppcheck warning/error cleanup

**System(s) tested o [... cut]
closed 2019-11-30 by skliper