nasa/osal #303: cppcheck warning misc. dir.

avan989, 2019-12-11:
**Is your feature request related to a problem? Please describe.**
cppcheck warning from #301 

```
[osal/src/bsp/mcf5235-rtems/src/bsp_start.c:136] -> [osal/src/bsp/mcf5235-rtems/src/bsp_start.c:143]: (style) Variable 'status' is reassigned a value before the old one has been used.
[osal/src/bsp/mcf5235-rtems/src/bsp_start.c:143] -> [osal/src/bsp/mcf5235-rtems/src/bsp_start.c:144]: (style) Variable 'status' is reassigned a value before the old one has been used.
[osal/src/bsp/mcf5235-rtems/src/bsp_start.c:144] -> [osal/src/bsp/mcf5235-rtems/src/bsp_start.c:145]: (style) Variable 'status' is reassigned a value before the old one has been used.
[osal/src/bsp/mcf5235-rtems/src/bsp_start.c:145]: (style) Variable 'status' is assigned a value that is never used.
[osal/src/bsp/sis-rtems/src/bsp_start.c:135] -> [osal/src/bsp/sis-rtems/src/bsp_start.c:142]: (style) Variable 'status' is reassigned a value before the old one has been used.
[osal/src/bsp/sis-rtems/src/bsp_start.c:142] -> [osal/src/bsp/sis-rtems/src/bsp_start.c:143]: (style) Variable 'status' is reassigned a value before the old one has been used.
[osal/src/bsp/sis-rtems/src/bsp_start.c:143] -> [osal/src/bsp/sis-rtems/src/bsp_start.c:144]: (style) Variable 'status' is reassigned a value before the old one has been used.
[osal/src/bsp/sis-rtems/src/bsp_start.c:144]: (style) Variable 'status' is assigned a value that is never used.
[osal/src/bsp/sparc-vxworks6.7/ut-src/bsp_ut.c:50]: (style) Unused variable: mode
[osal/src/bsp/sparc-vxworks6.7/ut-src/bsp_ut.c:216]: (style) Variable 'status' is assigned a value that is never used.
[osal/src/os/portable/os-impl-posix-dl.c:54]: (style) struct member 'OS_impl_module_internal_record_t::dl_handle' is never used.
[osal/src/os/posix/os-posix.h:110] -> [osal/src/os/posix/osapi.c:737]: (style, inconclusive) Function 'OS_Posix_InternalTaskCreate_Impl' argument 1 names different: declaration 'thr' definition 'pthr'.
[osal/src/os/posix/os-posix.h:110] -> [osal/src/os/posix/osapi.c:737]: (style, inconclusive) Function 'OS_Posix_InternalTaskCreate_Impl' argument 4 names different: declaration 'Entry' definition 'entry'.
[osal/src/os/posix/osapi.c:163] -> [osal/src/os/posix/osapi.c:2264]: (style, inconclusive) Function 'OS_CompAbsDelayTime' argument 1 names different: declaration 'milli_second' definition 'msecs'.
[osal/src/os/rtems/osloader.c:43]: (style) struct member 'OS_impl_module_internal_record_t::dl_handle' is never used.
[osal/src/tests/mutex-test/mutex-test.c:46] -> [osal/src/tests/mutex-test/mutex-test.c:47]: (style) Variable 'status' is reassigned a value before the old one has been used.
[osal/src/tests/mutex-test/mutex-test.c:98] -> [osal/src/tests/mutex-test/mutex-test.c:100]: (style) Variable 'status' is reassigned a value before the old one has been used.
[osal/src/tests/mutex-test/mutex-test.c:153] -> [osal/src/tests/mutex-test/mutex-test.c:154]: (style) Variable 'status' is reassigned a value before the old one has been used.
```

**Describe the solution you'd like**
resolve cppcheck warning.

**Requester Info**
Anh Van, NASA Goddard

jphickey, 2019-12-11:
Note that there is currently no way to test/verify changes related to the "sis-rtems", "mcf5235-rtems", or "sparc-vxworks6.7" bsp code.  

As there is no corresponding CFE PSP for these, we might just want to drop them.  At least the two RTEMS ones are almost certainly stale and haven't been verified in years, and the sparc-vxworks platform may not be current either.

skliper, 2019-12-11:
See #307, no need to fix warnings in those BSPs.

closed 2019-12-12 by avan989