nasa/osal #304: cppcheck warning unit-test

avan989, 2019-12-11:
**Is your feature request related to a problem? Please describe.**
cppcheck warning from #301 

```
[osal/src/unit-tests/osfile-test/ut_osfile_test.c:62] -> [osal/src/unit-tests/osfile-test/ut_osfile_test.c:64]: (style) Variable 'res' is reassigned a value before the old one has been used.
[osal/src/unit-tests/osfile-test/ut_osfile_test_arinc653.c:70] -> [osal/src/unit-tests/osfile-test/ut_osfile_test_arinc653.c:73]: (style) Variable 'res' is reassigned a value before the old one has been used.
[osal/src/unit-tests/osloader-test/ut_module.c:5]: (style) Variable 'i' is assigned a value that is never used.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:92]: (warning) %d in format string (no. 4) requires 'int' but the argument type is 'signed long'.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:92]: (warning) %d in format string (no. 6) requires 'int' but the argument type is 'unsigned int'.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:92]: (warning) %d in format string (no. 8) requires 'int' but the argument type is 'unsigned int'.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:105]: (warning) %d in format string (no. 4) requires 'int' but the argument type is 'signed long'.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:105]: (warning) %d in format string (no. 6) requires 'int' but the argument type is 'unsigned int'.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:105]: (warning) %d in format string (no. 8) requires 'int' but the argument type is 'unsigned int'.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:153] -> [osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:170]: (style) Variable 'test_int' is reassigned a value before the old one has been used.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:154] -> [osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:171]: (style) Variable 'test_long' is reassigned a value before the old one has been used.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:155] -> [osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:172]: (style) Variable 'test_unsigned' is reassigned a value before the old one has been used.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:156] -> [osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:173]: (style) Variable 'test_hex' is reassigned a value before the old one has been used.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:159] -> [osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:176]: (style) Variable 'test_char' is reassigned a value before the old one has been used.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:170]: (style) Variable 'test_int' is assigned a value that is never used.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:171]: (style) Variable 'test_long' is assigned a value that is never used.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:172]: (style) Variable 'test_unsigned' is assigned a value that is never used.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:173]: (style) Variable 'test_hex' is assigned a value that is never used.
[osal/src/unit-tests/osprintf-test/ut_osprintf_misc.c:176]: (style) Variable 'test_char' is assigned a value that is never used.
[osal/src/unit-tests/shared/ut_os_stubs.c:116] -> [osal/src/unit-tests/shared/ut_os_stubs.c:120]: (style) Variable 'apiInfo' is reassigned a value before the old one has been used.
[osal/src/unit-tests/shared/ut_os_stubs.h:171] -> [osal/src/unit-tests/shared/ut_os_stubs.c:159]: (style, inconclusive) Function 'UT_os_set_log_verbose' argument 1 names different: declaration 'verboseLevel' definition 'vLevel'.
```

**Describe the solution you'd like**
resolve cppcheck warning

**Requester Info**
Anh Van, NASA Goddard

jphickey, 2019-12-11:
I'm not sure it is worth investing the time to fix minor style issues in this test code at this time.  This code needs a complete revamp and conversion to UT assert style to be consistent with the other tests.

This code also "hides" a lot of its logic in macros which I believe is against other coding style guidelines so that should be fixed first.

skliper, 2019-12-11:
@avan989 Priority is enforcing no warnings on the flight code
- osal/src/bsp
- osal/src/os

skliper, 2020-01-02:
No plan to fix style issues in test code.

closed 2020-01-02 by skliper