nasa/osal #309: cppcheck test dir.

avan989, 2019-12-12:
**Is your feature request related to a problem? Please describe.**
From issue #303 

```
[osal/src/tests/mutex-test/mutex-test.c:46] -> [osal/src/tests/mutex-test/mutex-test.c:47]: (style) Variable 'status' is reassigned a value before the old one has been used.
[osal/src/tests/mutex-test/mutex-test.c:98] -> [osal/src/tests/mutex-test/mutex-test.c:100]: (style) Variable 'status' is reassigned a value before the old one has been used.
[osal/src/tests/mutex-test/mutex-test.c:153] -> [osal/src/tests/mutex-test/mutex-test.c:154]: (style) Variable 'status' is reassigned a value before the old one has been used.
```

**Requester Info**
Anh Van, NASA Goddard

jphickey, 2019-12-12:
See also #157

skliper, 2020-01-02:
No plan to fix cppcheck "style" issues in test code.

closed 2020-01-02 by skliper