nasa/osal #327: Recommended osal issues

avan989, 2019-12-20:
**Is your feature request related to a problem? Please describe.**
Recommended osal issues:

osapi-os-core.h
```
line 50:
/* #define for enabling floating point operations on a task*/
--
  | This comment appears to contain commented-out code
```

osapi-module.c
```
line 319:
/* module_prop->creator = record->creator; */
--
  | This comment appears to contain commented-out code

```

shared/os-impl.h
```
typedef enum
--
367 | {
  | In an enumerator list, the = construct should not be used to explicitly initialize members other than the first, unless all items are explicitly initialized.
368 | OS_FILE_FLAG_NONE,
369 | OS_FILE_FLAG_CREATE = 0x01,
```

osal/src/os/posix/osfileapi.c
```
*/
--
81 | #include "../portable/os-impl-posix-io.c"
  | The #include pre-processor directive should only be used to include header files.

*/
88 | #include "../portable/os-impl-posix-files.c"
  | The #include pre-processor directive should only be used to include header files.
89

*/
--
95 | #include "../portable/os-impl-posix-dirs.c"
  | The #include pre-processor directive should only be used to include header files.
```
posix/osapi.c:
```
/* use the portable version of OS_ConsoleWrite_Impl() */
--
2411 | #include "../portable/os-impl-console-directwrite.c"
  | The #include pre-processor directive should only be used to include header files.
```
posix/osloader.c
```
*/
--
32 | #include "../portable/os-impl-posix-dl.c"
  | The #include pre-processor directive should only be used to include header files.
```

posix/osnetwork.c
```
*/
--
58 | #include "../portable/os-impl-bsd-sockets.c"
  | The #include pre-processor directive should only be used to include header files.
```

posix/osselect.c
```
/* Leverage the portable UNIX-style Select API for POSIX */
--
35 | #include "../portable/os-impl-bsd-select.c"
  | The #include pre-processor directive should only be used to include header files.
```
posix/ostimer.c
```
/* POSIX implements clock_gettime and clock_settime that can be used */
--
566 | #include "../portable/os-impl-posix-gettime.c"
  | The #include pre-processor directive should only be used to include header files.
```
posix/osnetwork.c
```
/*
--
25 | * FIXME: this belongs in the osconfig.h file, really.
26 | */
  | FIXME comment: this belongs in the osconfig.h file, really.
27 | #define OS_NETWORK_SUPPORTS_IPV6
```

shared/osapi-select.c
```
/*
--
85 | * FIXME:
86 | * This does not currently increment any refcounts.
87 | * That means a file/socket can be closed while actively inside a
88 | * OS_SelectMultiple() call in another thread.
89 | */
  | FIXME comment
```

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context about the feature request here.

**Requester Info**
Full name and company/organization if applicable

skliper, 2019-12-22:
Moved #include of c files to #285

skliper, 2020-10-22:
FIXME moved to #630

PR #633 (2020-11-02): Fix #327 and #333, Squash LGTM warnings
**Describe the contribution**
- Turned all FIXME comments into issues (and removed FIXME)
- Fix #333: security/filename race issue resolved by opening file and acting on descriptor (had to add fstat stub)
- Fix #327: squashed the minor recommended bugs

**Testing performed**
Built and ran unit tests, passed.

**Expected behavior changes**
None

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: bundle main + this commit

**Additional context**
#630, #632 - new issues from FIXME
#631 - uncovered missing functional test on [... cut]
commit 0d706ca5 (2020-11-02): Merge pull request #633 from skliper/fix327-lgtm-recommended

Fix #327 and #333, Squash LGTM warnings
commit f2efc1ca (2020-11-02): Fix #327 and #333, Squash LGTM warnings
PR #639 (2020-11-04): Integration Candidate: 2020-11-03
**Describe the contribution**

Fix #327 and Fix #333, Squash LGTM warnings
Fix #623, add debug messages for mutex double locks
Fix #628, Update UtAssert macros with dynamic string formatting

**Testing performed**
See framework-bundle CI: https://github.com/nasa/cFS/pull/154/checks

**Expected behavior changes**

PR #634 - When `OS_DEBUG` is enabled, this adds a message if mutex give/take actions occur outside the expected sequence. This informs the user (via the debug console) if a lock is taken more than once or if a lock is given by a different task than th [... cut]
closed 2020-11-04 by astrogeco