nasa/osal #329: lgtm warning - misc.

avan989, 2019-12-23:
**Describe the bug**
reference issues #320 

os-impl-posix-files.c

```
line 240:

if ( chmod(local_path, st.st_mode) < 0 )
--
  | The filename being operated upon was previously checked, but the underlying file may have been changed since then.
```

posix/osfileapi.c

```
line 186:
execl(shell, "sh", "-c", Cmd, NULL); /* does not return if successful */
--
  | The value of this argument may come from call to getenv and is being passed to execl
```

osapi-filesys.c

```
line 330:
OS_FileSysStopVolume_Impl(local_id);
--
  | This expression has no effect (because OS_FileSysStopVolume_Impl has no external side effects).
```

**Reporter Info**
Anh Van, NASA Goddard

avan989, 2019-12-26:
Moved osapi-filesys.c, and os-impl-posix-files.c to separate issue.

skliper, 2019-12-30:
This is a significant security concern.  Suggestion is to separate out the shell functionality and make it optionally included (preference is per build system option).  Or remove shell functionality entirely.

Related to https://github.com/nasa/cFE/issues/84

skliper, 2020-01-08:
CCB 20190108 - see comment on #332, will move forward with current pull request and consider separation/optional inclusion in the future.

PR #355 (2020-01-27): Integration Candidate 20200121
**Describe the contribution**
Fix #328, fix #329

**Testing performed**
Steps taken to test the contribution:
1. Current CI passes
1. Enhanced CI in #40 passes debug/with deprecated elements testing

**Expected behavior changes**
Resolved LGTM errors, no functional changes

**System(s) tested on**
 - CI: Ubuntu 18.04

**Additional context**
N/A

**Third party code**
N/A

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GSFC
closed 2020-01-27 by skliper