nasa/osal #333: lgtm issues for os-impl-posix-files.c, osapi-filesys.c

avan989, 2019-12-26:
**Is your feature request related to a problem? Please describe.**
os-impl-posix-files.c

```
line 240:

if ( chmod(local_path, st.st_mode) < 0 )
--
  | The filename being operated upon was previously checked, but the underlying file may have been changed since then.
```

osapi-filesys.c

```
line 330:
OS_FileSysStopVolume_Impl(local_id);
--
  | This expression has no effect (because OS_FileSysStopVolume_Impl has no external side effects).
```

**Requester Info**
Anh Van, NASA Goddard

avan989, 2019-12-26:
Reference https://github.com/nasa/osal/pull/326:

os-impl-posix-files.c: 
possible solution use open() to get file descriptor with O_PATH flag but it is Linux-specific (non-posix).

osapi-filesys.c:
On second look maybe the OS_FileSysStopVolume_Impl shouldn't be entirely empty here.

closed 2020-11-04 by astrogeco

jphickey, 2020-11-13:
FYI - turns out this PR broke our VxWorks build:

```
/home/jphickey/cfecfs/github/osal/src/os/portable/os-impl-posix-files.c: In function 'OS_FileChmod_Impl':
/home/jphickey/cfecfs/github/osal/src/os/portable/os-impl-posix-files.c:202: error: too few arguments to function 'open'
make[7]: *** [osal/vxworks_impl/CMakeFiles/osal_vxworks_impl.dir/__/portable/os-impl-posix-files.c.obj] Error 1
make[6]: *** [osal/vxworks_impl/CMakeFiles/osal_vxworks_impl.dir/all] Error 2
```

PR #633 (2020-11-02): Fix #327 and #333, Squash LGTM warnings
**Describe the contribution**
- Turned all FIXME comments into issues (and removed FIXME)
- Fix #333: security/filename race issue resolved by opening file and acting on descriptor (had to add fstat stub)
- Fix #327: squashed the minor recommended bugs

**Testing performed**
Built and ran unit tests, passed.

**Expected behavior changes**
None

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: bundle main + this commit

**Additional context**
#630, #632 - new issues from FIXME
#631 - uncovered missing functional test on [... cut]
PR #639 (2020-11-04): Integration Candidate: 2020-11-03
**Describe the contribution**

Fix #327 and Fix #333, Squash LGTM warnings
Fix #623, add debug messages for mutex double locks
Fix #628, Update UtAssert macros with dynamic string formatting

**Testing performed**
See framework-bundle CI: https://github.com/nasa/cFS/pull/154/checks

**Expected behavior changes**

PR #634 - When `OS_DEBUG` is enabled, this adds a message if mutex give/take actions occur outside the expected sequence. This informs the user (via the debug console) if a lock is taken more than once or if a lock is given by a different task than th [... cut]