nasa/osal #335: osal_timer_UT periodic failure

skliper, 2019-12-31:
**Describe the bug**
osal_timer_UT periodically reports a failure on linux for the nominal OS_TimerSet case

**To Reproduce**
1. Build and run
```
make ENABLE_UNIT_TESTS=TRUE SIMULATION=native prep
make
make install
build/exe/cpu1/osal_timer_UT
```

**Expected behavior**
Should pass

**Code snips**
See ut_ostimer_test.c for UT_os_timercallback (the timer callback function)
See ut_ostimer_timerio_test.c line 654 for the failure location

**System observed on:**
 - cFS Dev 2 Server
 - OS: Ubuntu 18.04
 - Versions: happened to be running the ic-20191230 branches, but not unique (historical issue)

**Additional context**
Added after the tolerance check in the callback function:
```
printf("\ndeltaTime = %d\n", deltaTime);
printf("g_toleranceVal = %d\n", g_toleranceVal);
printf("currIntervalTime = %d\n", currIntervalTime);
printf("prevIntervalTime = %d\n", prevIntervalTime);
printf("endTime = %d sec, %d msec\n", endTime.seconds, endTime.microsecs);
printf("currTime = %d sec, %d msec\n", currTime.seconds, currTime.microsecs);
```
failure result was seen due to unexpected first interval of 1056, where interval was set to 500000:
```
deltaTime = 0
g_toleranceVal = 25000
currIntervalTime = 0
prevIntervalTime = 0
endTime = 1577812468 sec, 179711 msec
currTime = 1577812468 sec, 179711 msec

deltaTime = 1056
g_toleranceVal = 25000
currIntervalTime = 1056
prevIntervalTime = 0
endTime = 1577812468 sec, 180767 msec
currTime = 1577812468 sec, 179711 msec

deltaTime = 498945
g_toleranceVal = 25000
currIntervalTime = 500001
prevIntervalTime = 1056
endTime = 1577812468 sec, 680768 msec
currTime = 1577812468 sec, 180767 msec

deltaTime = 12
g_toleranceVal = 25000
currIntervalTime = 499989
prevIntervalTime = 500001
endTime = 1577812469 sec, 180757 msec
currTime = 1577812468 sec, 680768 msec
```
**Reporter Info**
Jacob Hageman - NASA/GSFC

skliper, 2019-12-31:
@jphickey any relation to the vxworks timer startup issue?

skliper, 2019-12-31:
Actual failure comes from the deltaTime = 498945 the 3rd time through, since the delta isn't checked on the first 2 frames.

skliper, 2019-12-31:
Now I'm confused.  The start time is set to 1000, so in this case the unit under test did what was expected.  I ran it a few more times and DID get it to pass, which is likely the actual error.

jphickey, 2019-12-31:
Can't really add printf statements to the timer callback function, as these can disrupt the timing substantially - so I'm not sure if that observation is really what is happening or if it is just an artifact of doing the printf's.

skliper, 2019-12-31:
The timer is slow, and there is plenty of time to print prior to next call.  They are after getting time.

skliper, 2019-12-31:
I understand it *could* effect it, but not worth bothering w/ a fancier test when the delay is .5 sec.  It does illustrate an issue in the test, so why the pushback?

skliper, 2019-12-31:
I'll fix the test for the first issue identified and remove the printf's to retest (since start delay of 1000 is likely impacted by the printfs).

skliper, 2019-12-31:
> Now I'm confused. The start time is set to 1000, so in this case the unit under test did what was expected. I ran it a few more times and DID get it to pass, which is likely the actual error.

Retracting this.  When it does pass, the first interval is ~500000 as expected.  Just a guess since for all the failure cases the first interval is ~1000 usec, but could it be the callback is inappropriately getting called before the start delay?  This doesn't sound like expected behavior, although the ~1000 usec could be a coincidence it's suspect that it matches the start delay.

jphickey, 2019-12-31:
I think this is coincidence ...  The 1000usec initial start delay is pretty tight for a desktop Linux kernel which typically has tick rate of 100Hz, so this is unlikely to be accurate but it doesn't check the first interval anyway.   I just ran the test a bunch of times and when it finally failed (after many successful runs) it was on the 10th interval, which according to the check was only 7us after the previous call.  I will look closer.

skliper, 2019-12-31:
could it be related to the following comments?

https://github.com/nasa/osal/blob/155e9ebcd6d1930890231a44237e6883d229d22c/src/os/shared/osapi-time.c#L358-L370

Maybe the spurious callback case the first time through, then gets called after the start wait.

skliper, 2019-12-31:
Although it looks like a dedicated timer to me, so doesn't quite fit description.

jphickey, 2019-12-31:
I think I know what the issue is....  

This test is not comparing the elapsed interval to the expected interval, but rather the interval that came before it.  So yeah, if the _first_ interval is short (i.e. 1000us), it invalidates the _second_ test which will end up comparing 1000us to 500000us and getting a difference of 499000us.

I'm somewhat surprised this doesn't fail more often than it does.

jphickey, 2019-12-31:
I'm currently testing a fix in a loop; if this runs repeatedly for an hour or so with no failures than I'll push it up.

skliper, 2020-01-02:
Note the real issue is described and fixed in #336.  In short - pending signal from previous test could cause an extra "spurious" initial callback at the start, now cleared at setup.

PR #336 (2020-01-08): Clear pending signal when configuring timer
**Describe the contribution**

Fix #335 

On POSIX, call sigtimedwait() on the selected RT signal as part of the set up for the timebase.  This ensures that if the signal is already pending, it will be cleared.

This also simplifies the timer callback routine in the UT code, cleaning up some unnecessary extra logic.

**Testing performed**
Build with ENABLE_UNIT_TESTS=TRUE and execute `osal_timer_UT` in a repeated loop.

Prior to this fix, it would typically fail after about 10-20 iterations, sometimes fewer.
After this fix, it does not fail (100+ ite [... cut]
commit 3bdea3a8 (2020-01-08): Fix #335: Clear pending signal when configuring timer

On POSIX, call sigtimedwait() on the selected RT signal as part of
the set up for the timebase.  This ensures that if the signal is
already pending, it will be cleared.

This also simplifies the timer callback routine in the UT code,
cleaning up some unncessary extra logic.
commit e9d9e1b6 (2020-01-13): Fix #335: Clear pending signal when configuring timer

On POSIX, call sigtimedwait() on the selected RT signal as part of
the set up for the timebase.  This ensures that if the signal is
already pending, it will be cleared.

This also simplifies the timer callback routine in the UT code,
cleaning up some unncessary extra logic.
commit 52bf8932 (2020-01-14): Fix #335: Clear pending signal when configuring timer

On POSIX, call sigtimedwait() on the selected RT signal as part of
the set up for the timebase.  This ensures that if the signal is
already pending, it will be cleared.

This also simplifies the timer callback routine in the UT code,
cleaning up some unncessary extra logic.
closed 2020-01-17 by skliper