nasa/osal #337: Possible bug in os/posix/osapi.c when running osal-core-test / BSEM tests.

stanislaw, 2020-01-02:
**Describe the bug**

I apologize in advance in case the issue I am reporting is caused by my misunderstanding of the OSAL POSIX code and the problem is rather somewhere on my end.

I am trying to port the POSIX OSAL code to macOS. It is a rather long adventure but it looks like there can be a happy end. By now I am going through all of the unit and integration tests in this repository in order to see which things have to be changed in order to run on macOS.

This issue I have first reported on StackOverflow because I thought the problem was on the macOS side: [Calling pthread_cond_destroy results in “Function not implemented” ENOSYS on macOS](https://stackoverflow.com/questions/59560940/calling-pthread-cond-destroy-results-in-function-not-implemented-enosys-on-mac).

```c
if (pthread_cond_destroy(&(sem->cv)) != 0) {
    printf("pthread_cond_destroy %d %s\n", errno, strerror(errno)); // my addition
    ...
}
```

results in `pthread_cond_destroy 78 Function not implemented` as I am running it on macOS.

Looking at the code further I have found that there might be a possible bug in the `OS_BinSemCreate_Impl` function along these lines:

```
        /*
         ** Initialize the condition variable
         */
        ret = pthread_cond_init(&(sem->cv), NULL);
        if (ret != 0)
        {
            OS_DEBUG("Error: pthread_cond_init failed: %s\n",strerror(ret));
            return_code = OS_SEM_FAILURE;
            break;
        }

        cond_created = 1;

        /*
         ** fill out the proper OSAL table fields
         */

        memset(sem, 0, sizeof (*sem));
        sem->current_value = initial_value;

        return_code = OS_SUCCESS;
```

I would like to highlight that the `memset(sem, 0, sizeof (*sem));` happens AFTER the `ret = pthread_cond_init(&(sem->cv), NULL);` line which means that `memset` corrupts the work that is done by `pthread_cond_init`. 

If I put the `memset` string BEFORE the `pthread_cond_init` line, the `BSEM` test passes.

**To Reproduce**

I am reproducing this on a private fork of `nasa/osal` and it is a very hacky branch to make it work on macOS. At the moment I can only suggest to do a mental reasoning about the critical lines: `memset...` and `pthread_cond_init...` lines as I described above.

**Expected behavior**

This is how the test log looks like with my change above and this is what I expect to always happen:

```
/Users/stanislaw/workspace/code/cfs-osal-posix-mac/cmake-build-debug/tests/osal-core-test

[BEGIN] PC-LINUX UNIT TEST

[BEGIN] 01 BSEM
[ PASS] 01.001 osal-core-test.c:312 - OS_BinSemCreate, nominal
[ PASS] 01.002 osal-core-test.c:312 - OS_BinSemCreate, nominal
[ PASS] 01.003 osal-core-test.c:312 - OS_BinSemCreate, nominal
[ PASS] 01.004 osal-core-test.c:312 - OS_BinSemCreate, nominal
[ PASS] 01.005 osal-core-test.c:312 - OS_BinSemCreate, nominal
[ PASS] 01.006 osal-core-test.c:312 - OS_BinSemCreate, nominal
[ PASS] 01.007 osal-core-test.c:312 - OS_BinSemCreate, nominal
[ PASS] 01.008 osal-core-test.c:312 - OS_BinSemCreate, nominal
[ PASS] 01.009 osal-core-test.c:312 - OS_BinSemCreate, nominal
[... 143 lines cut ...]
[ FAIL] 01.045 osal-core-test.c:342 - OS_BinSemCreate, recreate 2
[ FAIL] 01.046 osal-core-test.c:346 - OS_BinSemCreate, recreate 3
[ PASS] 01.047 osal-core-test.c:353 - OS_BinSemGetIdByName, Bin 0
[ FAIL] 01.048 osal-core-test.c:357 - OS_BinSemGetIdByName, Bin 1
[ PASS] 01.049 osal-core-test.c:361 - OS_BinSemGetIdByName, Bin 2
[ PASS] 01.050 osal-core-test.c:365 - OS_BinSemGetIdByName, Bin 3
pthread_cond_destroy 78 Function not implemented
[ PASS] 01.051 osal-core-test.c:371 - OS_BinSemDelete, Old ID
pthread_cond_destroy 78 Function not implemented
[ FAIL] 01.052 osal-core-test.c:375 - OS_BinSemDelete, Bin 0
pthread_cond_destroy 78 Function not implemented
[ PASS] 01.053 osal-core-test.c:380 - OS_BinSemDelete, Bin 1
pthread_cond_destroy 78 Function not implemented
[ FAIL] 01.054 osal-core-test.c:383 - OS_BinSemDelete, Bin 2
pthread_cond_destroy 78 Function not implemented
[ FAIL] 01.055 osal-core-test.c:386 - OS_BinSemDelete, Bin 3
[  END] 01 BSEM                 TOTAL::55    PASS::28    FAIL::27     MIR::0      TSF::0      N/A::0   

COMPLETE: 1 tests Segment(s) executed

Process finished with exit code 2
```

**Reporter Info**
Stanislav Pankevich

p-po, 2020-01-02:
Looks like it originated from this commit bfa7a33020e9a77221352a99bd2dbee4860ba852.

jphickey, 2020-01-02:
You are correct, the memset absolutely should _not_ be way down there, it should be near the top of the function.  I'm not sure how it ended up there (might have been a bad merge that was never caught way back when).

My recommendation is to put the memset immediately after the `sem` pointer is assigned (e.g. Line 1467).

Thank you for catching and reporting this.  Can you submit a pull request with this change?

I'd also be curious to find out any other modifications you needed to make in order to get OSX to work.

stanislaw, 2020-01-02:
Thanks for the quick answer and the confirmation. I am preparing a PR.

stanislaw, 2020-01-04:
> I'd also be curious to find out any other modifications you needed to make in order to get OSX to work.

This is the work that has to be done:

1) mqueue is missing on macOS (advanced, but mostly done) 

By now I am missing the `mq_timedreceive` and `mq_timedsend` functions. I am reusing the implementation based on memory-mapped files from the book by Stevens as found [here](https://github.com/shihyu/Linux_Programming/blob/master/books/W.%20Richard%20Stevens%20-%20Unix%20Network%20Programming%20Vol2.pdf). I have recovered the source code attached to the book and put it to a modern CMake environment with all of the author's original tests converted to separate Google Test cases.

1a) `sigqueue` is missing on macOS but it looks like OSAL does not use the `_nofify` functionality of `mqueue` so it should not be a problem.

2) `clock_nanosleep` (easy, done)

This call is just a convenience method that calls into `nanosleep`.

3) `timer_create`, `timer_settime`, `timer_delete` are missing on macOS (advanced). Right now I am studying how `ostimer.c` works using `timer_*` functions, a private thread that is doing `sigwait` etc. I am under impression that it might be easier to not reimplement `timer_*` on macOS but rather use some other mechanism such as the timers from `libdispatch`.

4) `sem_timedwait` (pending, should be doable)

5) `pthread_setschedprio` is missing (pending but I am pretty sure one should be able to set priorities on macOS)

6) Some minor changes doable with `#ifdef __APPLE__` in two or three places (done)

My vacation days are ending soon so my progress is slowing down but I am quite committed to making the whole thing work. I would like to implement the items 3, 4 and 5 and then suggest a WIP Pull Request to open it for review (there are and will be some open questions where to put code).

---

This is a current progress. All uncommented tests are passing.

```makefile
# $ cat Makefile

test: test.unit test.integration

test.unit: build.cmake
	# cd build.commandline.dir && ./unit-tests/oscore-test/osal_core_UT
	cd build.commandline.dir && ./unit-tests/osfile-test/osal_file_UT
	cd build.commandline.dir && ./unit-tests/osfilesys-test/osal_filesys_UT
	# cd build.commandline.dir && ./unit-tests/osloader-test/osal_loader_UT
	cd build.commandline.dir && ./unit-tests/osnetwork-test/osal_network_UT

	# Working on this one:
	cd build.commandline.dir && ./unit-tests/ostimer-test/osal_timer_UT 

test.integration: build.cmake
	cd build.commandline.dir && ./tests/bin-sem-flush-test
	# cd build.commandline.dir && ./tests/bin-sem-test
	# cd build.commandline.dir && ./tests/bin-sem-timeout-test
	# cd build.commandline.dir && ./tests/count-sem-test
	cd build.commandline.dir && ./tests/file-api-test
	cd build.commandline.dir && ./tests/mutex-test
	cd build.commandline.dir && ./tests/osal-core-test
	# cd build.commandline.dir && ./tests/queue-timeout-test
	# cd build.commandline.dir && ./tests/sem-speed-test
	cd build.commandline.dir && ./tests/symbol-api-test
	# cd build.commandline.dir && ./tests/timer-test

build.cmake:
	mkdir -p build.commandline.dir
	cd build.commandline.dir && cmake -G Ninja -DOS_STATIC_SYMBOL_TABLE=0 -DENABLE_UNIT_TESTS=1 -DOSAL_SYSTEM_OSTYPE=posix -DOSAL_SYSTEM_BSPTYPE=pc-linux ..
	cd build.commandline.dir && ninja
```

---

Meanwhile I think there is a small issue that breaks some of the tests. Could you please confirm that it is intended like that? Currently in the master branch, the `OS_API_Init` is done after the call to `UT_os_timerinit_test` which makes some of its underlying code fail. If I change the order like on the diff, I get better results. The tests are still failing because I don't have `timer_*` implemented but that's definitely another error.

```diff
diff --git a/src/unit-tests/ostimer-test/ut_ostimer_test.c b/src/unit-tests/ostimer-test/ut_ostimer_test.c
index 709f56c..a944c7f 100644
--- a/src/unit-tests/ostimer-test/ut_ostimer_test.c
+++ b/src/unit-tests/ostimer-test/ut_ostimer_test.c
@@ -229,10 +229,10 @@ void OS_Application_Startup(void)

     UT_OS_LOG_MACRO("OSAL Unit Test Output File for ostimer APIs\n");

-    UT_os_timerinit_test();
-
     OS_API_Init();

+    UT_os_timerinit_test();
+
     /*
     ** Create the test task.
     */
```

jphickey, 2020-01-04:
> This is the work that has to be done:
> 
> 1. mqueue is missing on macOS (advanced, but mostly done)
> 
> By now I am missing the `mq_timedreceive` and `mq_timedsend` functions. I am reusing the 

Note that since (currently) all CFE/OSAL tasks run within the same memory space, there is really no need to use IPC-capable mechanism here, and that OSAL only uses a minimal set of mqueue features anyway.  

It has been suggested in the past that OSAL just provide its own mqueue by allocating a block of memory and using read/write pointers and local semaphore/mutex protection.  This could be a valuable fallback for any OS that doesn't provide a native queue API.

> 1a) `sigqueue` is missing on macOS but it looks like OSAL does not use the `_nofify` functionality of `mqueue` so it should not be a problem.

Correct, this is not needed.

> 1. `clock_nanosleep` (easy, done)
> 
> This call is just a convenience method that calls into `nanosleep`.

Yes, nanosleep is OK but note that the POSIX "clock_nanosleep" has a clock ID parameter which nanosleep assumes CLOCK_REALTIME.  The difference between these has to do if the clock is reset/changed during runtime.

Note that at least on on Linux the "clock_nanosleep" is only available if you specify the appropriate level of POSIX conformance.  MacOS might have a similar requirement.  As this is a POSIX-specified function I'd be surprised if it did not exist at all.

> 1. `timer_create`, `timer_settime`, `timer_delete` are missing on macOS (advanced). Right now I am studying how `ostimer.c` works using `timer_*` functions, a private thread that is doing `sigwait` etc. I am under impression that it might be easier to not reimplement `timer_*` on macOS but rather use some other mechanism such as the timers from `libdispatch`.

Note that in a previous OSAL revision, there was the ability to not use signals at all here but rather mimic the effect by running "clock_nanosleep" and repeatedly polling the system time to get the desired effect.  Might be worth looking into reviving this feature?

That being said, these are POSIX functions so they might be hidden behind a conformance level switch as well.

> 3. `pthread_setschedprio` is missing (pending but I am pretty sure one should be able to set priorities on macOS)

I noticed BSD is also missing this function.  But there is a setchedparam that sets both priority and scheduling mode that might be a workaround.  (this would be a good change as it improves BSD support too).
 
Thanks for the feedback!  As always I'm interested in making this as portable as possible so any feedback/testing to improve cross platform support like this is very helpful.

stanislaw, 2020-01-09:
Please see my answers below. Also find the current progress report after my answers.

> Note that since (currently) all CFE/OSAL tasks run within the same memory space, there is really no need to use IPC-capable mechanism here, and that OSAL only uses a minimal set of mqueue features [...]

I had this in mind (CFS should not do anything IPC in mqueue) but at the same time I thought it was a good idea to recover the original Steven's implementation of mqueue because I have never worked with mqueue API myself. Whether OSAL is going to implement this pure memory-based solution or not, the Steven's implementation seems to work and there is some useful bits I am learning from it. And it seems to be working in the context of nasa/osal which is most important for me at the moment.

> > 1a) `sigqueue` is missing on macOS but it looks like OSAL does not use the `_nofify` functionality of `mqueue` so it should not be a problem.
> 
> Correct, this is not needed.

Good, thanks.

> Yes, nanosleep is OK but note that the POSIX "clock_nanosleep" has a clock ID parameter which nanosleep assumes CLOCK_REALTIME. The difference between these has to do if the clock is reset/changed [...]

I am not sure if I know how to change this level of POSIX conformance on macOS. Have never heard of it. One indirect proof of its absence could be that it is missing in the Apple open source implementation while `nanosleep` [is there](https://opensource.apple.com/tarballs/Libc/). Thanks for the hint, I think it is definitely worth to revisit the available Apple implementation and implement something more advanced for `clock_nanosleep`. 

> Note that in a previous OSAL revision, there was the ability to not use signals at all here but rather mimic the effect by running "clock_nanosleep" and repeatedly polling the system time to get the [...]

The same. Could not find anything that would hint at their presence on macOS. I ended up implementing these functions using dispatch timers. And instead of `sigwait` (RT signals do not exist on macOS and `-rt` does not exit), I am doing

```  
//    ret = sigwait(&local->sigset, &sig);
    ret = timer_poll(local->host_timerid);
```

which imitates reading accumulated signals by reading a counter. It is extremely hacky but it makes things pass. Later it should be made a subject of next iteration.

> > 1. `pthread_setschedprio` is missing (pending but I am pretty sure one should be able to set priorities on macOS)
> 
> I noticed BSD is also missing this function. But there is a setchedparam that sets both priority and scheduling mode that might be a workaround. (this would be a good change as it improves BSD support too).

Yes, `setchedparam` works. Let's see how exactly it can be integrated.

I also forgot to mention that not only `sem_timedwait` is missing on macOS but also the related symbols `sem_init`, `sem_destroy`, etc, for unnamed semaphores are deprecated. I have found a version based on `pthread_cond_timedwait` on StackOverflow and with some minor modifications there is an implementation using symbols `mac_sem_*`. 

> Thanks for the feedback! As always I'm interested in making this as portable as possible so any feedback/testing to improve cross platform support like this is very helpful.

For me this is quite important to make CFS run on macOS because it's been my development system for a few years now. Just don't want to be locked to Linux at work only because of this OSAL limitation. So it is worth some spare time effort because at work I am setting up a Linux machine already :)

---

By now all of the tests are passing. The next step is to run the whole CFS on top of my posix-mac branch to see if there is anything visibly broken. I hope that it should run and after that and after some cleanups I could open that branch for review and further iterations.

```
test: test.unit test.integration

test.unit: build.cmake
	cd build.commandline.dir && ./unit-tests/oscore-test/osal_core_UT
	cd build.commandline.dir && ./unit-tests/osfile-test/osal_file_UT
	cd build.commandline.dir && ./unit-tests/osfilesys-test/osal_filesys_UT
	cd build.commandline.dir && ./unit-tests/osloader-test/osal_loader_UT
	cd build.commandline.dir && ./unit-tests/ostimer-test/osal_timer_UT
	cd build.commandline.dir && ./unit-tests/osnetwork-test/osal_network_UT

test.integration: build.cmake
	cd build.commandline.dir && ./tests/bin-sem-flush-test
	cd build.commandline.dir && ./tests/bin-sem-test
	cd build.commandline.dir && ./tests/bin-sem-timeout-test
	cd build.commandline.dir && ./tests/count-sem-test
	cd build.commandline.dir && ./tests/file-api-test
	cd build.commandline.dir && ./tests/mutex-test
	cd build.commandline.dir && ./tests/osal-core-test
	cd build.commandline.dir && ./tests/queue-timeout-test
	cd build.commandline.dir && ./tests/sem-speed-test
	cd build.commandline.dir && ./tests/symbol-api-test
	cd build.commandline.dir && ./tests/timer-test

build.cmake:
	mkdir -p build.commandline.dir
	cd build.commandline.dir && cmake -G Ninja -DOS_STATIC_SYMBOL_TABLE=0 -DENABLE_UNIT_TESTS=1 -DOSAL_SYSTEM_OSTYPE=posix -DOSAL_SYSTEM_BSPTYPE=pc-linux ..
	cd build.commandline.dir && ninja
```

stanislaw, 2020-01-12:
This is another progress update. By now I am running cFS without obvious crashes on macOS (the log is below).

I had to do the changes from https://github.com/nasa/cFE/pull/474 and https://github.com/nasa/cFE/issues/476. And stub this for now:

```c
/******************************************************************************
**  Function: CFE_PSP_GetCFETextSegmentInfo
**
**  Purpose:
**    This function returns the start and end address of the CFE text segment.
**     It may not be implemented on all architectures.
...
*/
int32 CFE_PSP_GetCFETextSegmentInfo(cpuaddr *PtrToCFESegment, uint32 *SizeOfCFESegment)
{
    return CFE_PSP_ERROR_NOT_IMPLEMENTED;
}
```

The only unimplemented bit known to me is to make `elf2cfetbl` to also work with Mach-O format (for now I am creating the table from Docker Ubuntu image). One question here just in case: could a Python script instead of a C program be an option here or you recommend me to follow the existing program and simply mirror the existing ELF implementation and create a Mach-O one?

P.S. I think I need to open an issue somewhere to track the further progress on my macOS-related work. Which place would you recommend? An issue on cFE's issue tracker? Or is there a better place?

Thanks.

---

macOS log:

```
Stanislaw@home:cFS (macos)*$ ./cmake-build-debug/cpu1/cpu1/core-cpu1
CFE_PSP: Default Reset Type = PO
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 42
CFE_PSP: Default CPU Name: cpu1
CFE_PSP: Starting the cFE with a POWER ON reset.
CFE_PSP: Clearing out CFE CDS Shared memory segment.

CFE_PSP: Clearing out CFE CDS Shared memory segment.
CFE_PSP: Clearing out CFE Reset Shared memory segment.

CFE_PSP: Clearing out CFE Reset Shared memory segment.
CFE_PSP: Clearing out CFE User Reserved Shared memory segment.

CFE_PSP: Clearing out CFE User Reserved Shared memory segment.
2030-011-20:08:58.71650 POWER ON RESET due to Power Cycle (Power Cycle).

2030-011-20:08:58.71656 ES Startup: CFE_ES_Main in EARLY_INIT state

2030-011-20:08:58.71650 POWER ON RESET due to Power Cycle (Power Cycle).
2030-011-20:08:58.71656 ES Startup: CFE_ES_Main in EARLY_INIT state
CFE_PSP: CFE_PSP_AttachExceptions Called

CFE_PSP: CFE_PSP_AttachExceptions Called
2030-011-20:08:58.71659 ES Startup: CFE_ES_Main entering CORE_STARTUP state

2030-011-20:08:58.71660 ES Startup: Starting Object Creation calls.

2030-011-20:08:58.71661 ES Startup: Calling CFE_ES_CDSEarlyInit

2030-011-20:08:58.71659 ES Startup: CFE_ES_Main entering CORE_STARTUP state
2030-011-20:08:58.71660 ES Startup: Starting Object Creation calls.
2030-011-20:08:58.71661 ES Startup: Calling CFE_ES_CDSEarlyInit
2030-011-20:08:58.71668 ES Startup: Calling CFE_EVS_EarlyInit

2030-011-20:08:58.71671 Event Log cleared following power-on reset

2030-011-20:08:58.71668 ES Startup: Calling CFE_EVS_EarlyInit
2030-011-20:08:58.71671 Event Log cleared following power-on reset
[... 66 lines cut ...]
1980-012-14:03:20.20453 ES Startup: Core App: CFE_TBL created. App ID: 4

1980-012-14:03:20.20453 ES Startup: Core App: CFE_TBL created. App ID: 4
EVS Port1 42/1/CFE_TBL 1: cFE TBL Initialized.  cFE Version 6.7.4.0

EVS Port1 42/1/CFE_TBL 1: cFE TBL Initialized.  cFE Version 6.7.4.0
1980-012-14:03:20.25492 ES Startup: Finished ES CreateObject table entries.

1980-012-14:03:20.25502 ES Startup: CFE_ES_Main entering CORE_READY state

1980-012-14:03:20.25492 ES Startup: Finished ES CreateObject table entries.
1980-012-14:03:20.25502 ES Startup: CFE_ES_Main entering CORE_READY state
1980-012-14:03:20.25516 ES Startup: Error, Can't Open ES App Startup file: /cf/cfe_es_startup.scr EC = 0xFFFFFFFF

1980-012-14:03:20.25518 ES Startup: CFE_ES_Main entering APPS_INIT state

1980-012-14:03:20.25519 ES Startup: CFE_ES_Main entering OPERATIONAL state

1980-012-14:03:20.25516 ES Startup: Error, Can't Open ES App Startup file: /cf/cfe_es_startup.scr EC = 0xFFFFFFFF
1980-012-14:03:20.25518 ES Startup: CFE_ES_Main entering APPS_INIT state
1980-012-14:03:20.25519 ES Startup: CFE_ES_Main entering OPERATIONAL state
EVS Port1 42/1/CFE_TIME 21: Stop FLYWHEEL

EVS Port1 42/1/CFE_TIME 21: Stop FLYWHEEL
```

jphickey, 2020-01-13:
There is absolutely no requirement to use `elf2cfetbl`.  This tool should have been deprecated and replaced long ago for a variety of reasons.

In a future version of CFE, the goal is to use some type of more intelligent Message/Table data structure management system (aka "data dictionary"), rather than using C structures for this.  So I wouldn't invest too much time into an elf2cfetbl replacement at this juncture.  Ideally the replacement for this tool would be tied into the data dictionary.

stanislaw, 2020-01-13:
> There is absolutely no requirement to use elf2cfetbl. This tool should have been deprecated and replaced long ago for a variety of reasons.

> In a future version of CFE, the goal is to use some type of more intelligent Message/Table data structure management system (aka "data dictionary"), rather than using C structures for this. So I [...]

Thanks for the quick warning. This is very good to know because it is one work item less. I will continue using the stubbed sample...tbl that I obtained from building cFS using Docker.

stanislaw, 2020-02-01:
> There is absolutely no requirement to use elf2cfetbl. This tool should have been deprecated and replaced long ago for a variety of reasons.

> In a future version of CFE, the goal is to use some type of more intelligent Message/Table data structure management system (aka "data dictionary"), rather than using C structures for this. So I [...]

I am getting some errors related to the `elf2cfetbl` tool on macOS which means something is not functioning properly. This is how I have commented out the calls to this tool:

```diff
-        COMMAND ${MISSION_BINARY_DIR}/tools/elf2cfetbl/elf2cfetbl ${TBLWE}.o
-        DEPENDS ${MISSION_BINARY_DIR}/tools/elf2cfetbl/elf2cfetbl ${TBL_SRC}
+        COMMAND echo "will apply elf tool to the ${TBLWE}.o"
+        COMMAND echo ${MISSION_BINARY_DIR}/bin/elf2cfetbl ${TBLWE}.o
+#        COMMAND ${MISSION_BINARY_DIR}/bin/elf2cfetbl ${TBLWE}.o
+#        DEPENDS ${MISSION_BINARY_DIR}/bin/elf2cfetbl ${TBL_SRC}
```

This is one example of what I am getting now.

```
1980-012-14:03:20.28436 ES Startup: SCH_LAB_APP loaded and created
open(./cf/sch_lab_table.tbl): No such file or directory
1980-012-14:03:20.28442 CFE_TBL:Load-App(8) Fail to load Tbl 'SCH_LAB_APP.SCH_LAB_SchTbl' from '/cf/sch_lab_table.tbl' (Stat=0xFFFFFFFF)
EVS Port1 42/1/CFE_TBL 93: SCH_LAB_APP Failed to Load 'SCH_LAB_APP.SCH_LAB_SchTbl' from '/cf/sch_lab_table.tbl', Status=0xFFFFFFFF
1980-012-14:03:20.28444 SCHEULE APP: Error Loading Table                                   SCH_LAB_SchTbl, RC = 0xFFFFFFFF
```

What would be your advice on how I could fix these errors so that the apps are loaded properly?

It feels like this is the last bit I need to overcome before I can open a WIP macOS-related PR to nasa/cfe.

jphickey, 2020-02-02:
Can you clarify the issue?  It looks like you've disabled the tool that generates the table file for the target, then noted that the table files failed to load on the target because the file is missing?

That error would be certainly expected, given that you have disabled the tool.  If you do not use `elf2cfetbl` then you need to generate the table file in some other manner.  It is a binary file that contains a struct that is loaded at runtime.  You can generate it with any tool/script of your choice (e.g. python?) but somehow you need to create this file if you want table loads to work.

stanislaw, 2020-02-02:
Sorry for the confusion! I do understand that the error would be certainly expected.

When I asked my question, I was basing it on the message that had I received from you earlier in this thread. Let me recover the full context:

> The only unimplemented bit known to me is to make elf2cfetbl to also work with Mach-O format (for now I am creating the table from Docker Ubuntu image). One question here just in case: could a [...]

> > > There is absolutely no requirement to use elf2cfetbl. This tool should have been deprecated and replaced long ago for a variety of reasons.

> In a future version of CFE, the goal is to use some type of more intelligent Message/Table data structure management system (aka "data dictionary"), rather than using C structures for this. **So I [...]

> > Thanks for the quick warning. This is very good to know because it is one work item less. I will continue using the stubbed sample...tbl that I obtained from building cFS using Docker.

> I am getting some errors related to the elf2cfetbl tool on macOS which means something is not functioning properly. This is how I have commented out the calls to this tool:

> This is one example of what I am getting now...

> What would be your advice on how I could fix these errors so that the apps are loaded properly?

> It feels like this is the last bit I need to overcome before I can open a WIP macOS-related PR to nasa/cfe.

So for me, it sounds like you want to get rid of this tool at some point or change the way these tables are created but for the time being I have to write a Mach-O equivalent for the ELF tool and I could write it using C or Python. Is that correct?

P.S. Sorry for not reading enough information from your comments. I still don't understand very well, which role this tool plays in having CFS/CFE to work properly.

jphickey, 2020-02-02:
The `elf2cfetbl` is merely one (example) way to create a binary table file.  With this tool, you write your table definition in C by creating a separate .c file which instantiates the table structure and initializes it to the data you want.  You then compile the file with the cross compiler, but rather than a normal link, the `elf2cfetbl` file extracts the structure definition content from the file, and writes it to a separate .tbl file with the right headers.

While this works, it is rather crude and only supports certain ELF-based tools (as you noted, MacOS apparently isn't one of them).  It also doesn't support more complex deployment scenarios, where table data might need to be extracted from a database or something like that.

When I said there is no requirement to use this specific tool, you _will_ need to somehow generate appropriate binary .tbl files for your target if you want to use apps that load runtime tables.  The format is pretty simple (C struct encapsulated with a FS+TBL header) so its not hard to hack out a python script to create these files, there is just no python tool already written AFAIK.

And yes, the goal is to put a better, more flexible, more cross-platform tool in the framework someday.  But that is probably at least a few releases away.

Before you spend the time writing your own tool, you might want to try simply invoking "objcopy" to see if you can convert your binary file to an ELF first, then run the tool against the copy.  It might work.  At least worth a shot I would think.

stanislaw, 2020-02-20:
> When I said there is no requirement to use this specific tool, you will need to somehow generate appropriate binary .tbl files for your target if you want to use apps that load runtime tables. The [...]

Please check out my branch where I have written a Python script. It does the job for both ELF and Mach-O files. The code is not in the final state and needs some cleanups but I think the high level details should be pretty clear.

I am using 3 Python packages: 

- `lief` for reading ELF and Mach-O files. See https://github.com/lief-project/LIEF.
- `lit` and `FileCheck` for integration tests.

If this direction is approved, I can open a PR to the `elf2cfetbl` repository for review.

https://github.com/nasa/elf2cfetbl/compare/master...stanislaw:macho2cfetbl?expand=1

Thank you for attention.

PR #338 (2020-02-25): Fix #337, fix memory corruption produced by misplaced memset()
**Describe the contribution**

This PR fixes #337. The exact change has been recommended by @jphickey.

**Testing performed**

Currently, I cannot test this on Linux only on macOS. As reported in the issue, I confirm that that the issue gets fixed with this change applied. Also, the bug has been confirmed by @jphickey and the changeset is created from his recommendation.

**Expected behavior changes**

See related issue: #337.

**System(s) tested on**

    Running osal-core-test tests (BSEM tests)
    OS: macOS Mojave 10.14.6 (18G8 [... cut]
commit ae68a3f1 (2020-02-25): Merge pull request #338 from stanislaw/fix-pthread-cond-init-issue

Fix #337, fix memory corruption produced by misplaced memset()
commit fe939281 (2020-02-25): Fix #337: fix memory corruption produced by misplaced memset()

Calling memset too late after assigning the `sem` variable in the code of
`OS_BinSemCreate_Impl` caused the data pointed to by `sem` to get corrupted.

The issue was not caught by the existing test suite for POSIX OSAL when running
on Linux. However running the test suite on macOS revealed the anomalies
in the behavior of `pthread_cond_destroy()` which was working on the corrupted
memory as was demonstrated in:

Calling pthread_cond_destroy results in “Function not implemented” ENOSYS on macOS,
https://stackoverflow.com/questions/ [... cut]
closed 2020-02-28 by astrogeco