nasa/osal #339: Remove classic build artifacts

skliper, 2020-01-02:
**Is your feature request related to a problem? Please describe.**
Classic build no longer supported.

**Describe the solution you'd like**
Remove all associated artifacts: build directory, all *.mak directories, setvars.sh

**Describe alternatives you've considered**
None

**Additional context**
Makes #164 obsolete

**Requester Info**
Jacob Hageman - NASA/GSFC

avan989, 2020-01-08:
I can do this.

CDKnightNASA, 2020-03-25:
Build instructions in doc/OSAL-Configuration-Guide.doc and .pdf have instructions for classic build. Also the section on building the unit tests requires use of the classic build. A new section needs to be written for building unit tests under cmake and the old classic build section needs to be removed (and refs to the classic build in the cmake section need to be stricken.)

skliper, 2020-03-25:
Preference is to open a new issue, see #384

commit 66cf5174 (2020-01-21): Fix #339, Remove classic build
commit b72031b8 (2020-01-21): Merge pull request #345 from avan989/remove_classic_build

Fix #339, Remove classic build
closed 2020-03-25 by skliper