nasa/osal #349: OS_SocketAccept fails

CDKnightNASA, 2020-01-13:
**Describe the bug**
When using OS_SocketAccept on a stream (TCP) server socket, it fails because the accept code calls OS_ObjectIdGetById() passing a OS_LOCK_MODE_REFCOUNT, but then OS_SocketAccept sees the refcount > 0 and fails out.

**To Reproduce**
Create code (e.g. an app) that creates a TCP socket server and try using OS_SocketAccept() to receive the incoming connections.

**Expected behavior**
OS_SocketAccept should work as expected/documented.

**Code snips**
OS_SockAddr_t Addr;
int Socket = 0, ClientFd;
int32 Status = 0;

OS_SocketAddrInit(&Addr, OS_SocketDomain_INET);
OS_SocketAddrFromString(Addr, "127.0.0.1");
OS_SocketOpen(&Socket, OS_SocketDomain_INET, OS_SocketType_STREAM);
OS_SocketBind(Socket, &Addr);
Status = OS_SocketAccept(Socket, &ClientFd, &Addr, 0);

**System observed on:**
Debian 9 64-bit x86

**Additional context**
removing the check for refcount results in expected behavior, OS accept() call should be thread-safe so need not maintain lock through SocketAccept() function.

**Reporter Info**
Chris Knight, NASA Ames

skliper, 2020-01-13:
Is it possible to create a functional test case to cover this?  The fact it doesn't work seems to indicate a missing functional test.

skliper, 2020-03-18:
@CDKnightNASA - any progress on this one?  Do you suggest we discuss at a CCB, or is there anything you need to help resolve it?

CDKnightNASA, 2020-03-19:
I need to write unit tests for it, hasn't been high on priority list. I can take a crack at it today.

CDKnightNASA, 2020-03-19:
FYI @jphickey there's a great dearth of unit tests for the network API...

So I wasn't clear after this last CCB whether we should use stubbed versions of the OS calls or use the real calls in unit testing. I am guessing that socket code should call the stubs as the OS may throw errors depending on firewall or other security-related configuration.

CDKnightNASA, 2020-03-19:
see #373

PR #372 (2020-04-01): Fix #349, OS_SocketAccept() fix
**Describe the contribution**
Fixes #349, note that this does *not* include unit tests. There don't seem to be any unit tests for the OSAL networking code. We should discuss at the CCB.

**Testing performed**
Tested with the TCP version of SBN, without the fix it fails on the Accept() call, with the fix, SBN nodes connect and traffic flows as expected.

**Expected behavior changes**
OS_SocketAccept() should work to accept new incoming TCP connections.

**System(s) tested on**
64-bit Debian 9 VM.

**Additional context**
Add any other context about the contributio [... cut]
commit b9ab1747 (2020-04-01): Merge pull request #372 from CDKnightNASA/fix-349-socket_accept

Fix #349, OS_SocketAccept() fix
PR #401 (2020-04-03): Integration Candidate 2020-03-25
**Describe the contribution**
Fixes #349

**Testing performed**
PR #372 
[Bundle CI](https://travis-ci.com/github/nasa/cfs/builds/158274344)

**Expected behavior changes**
 OS_SocketAccept() should work to accept new incoming TCP connections.

**System(s) tested on**
Bundle CI - Ubuntu: Bionic

**Contributor Info - All information REQUIRED for consideration of pull request**
Christopher. D. Knight - NASA/ARC
Gerardo E. Cruz-Ortiz - NASA/GSFC
closed 2020-04-03 by astrogeco