nasa/osal #351: Build Error: os-impl.h: No such file or directory

waheedejaz, 2020-01-15:
**Describe the bug**
Followed the build steps as in the README file but the build is not successfull. 

Error: 
`waheeed@ubuntu:~/Lifi/osal/build$ make
make -C tests 
make[1]: Entering directory '/home/waheeed/Lifi/osal/build/tests'
make -C bin-sem-flush-test 
make[2]: Entering directory '/home/waheeed/Lifi/osal/build/tests/bin-sem-flush-test'
gcc  -I/home/waheeed/Lifi/osal/src/inc -I/home/waheeed/Lifi/osal/src/os/inc -I/home/waheeed/Lifi/osal/src/tests/bin-sem-flush-test -I/home/waheeed/Lifi/osal/src/../ut_assert/inc -I../../inc   -D_EL -DENDIAN=_EL -DSOFTWARE_LITTLE_BIT_ORDER  -D__ix86__ -D_ix86_ -D_LINUX_OS_ -Dposix -DX86PC -DBUILD= -D_REENTRANT -D _EMBED_  -D_XOPEN_SOURCE=600   -Wall -pedantic -std=c99 -Wstrict-prototypes -O1 -DOS_DEBUG_LEVEL=3 -c -o osapi.o /home/waheeed/Lifi/osal/src/os/posix/osapi.c
In file included from /home/waheeed/Lifi/osal/src/os/posix/osapi.c:27:0:
/home/waheeed/Lifi/osal/src/os/posix/os-posix.h:42:10: fatal error: os-impl.h: No such file or directory
 #include "os-impl.h"
          ^~~~~~~~~~~
compilation terminated.`

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the latest build 5.3 (15-Jan-2020 )or the commit 781992ce2c0df067ebd84779615bf1ebf6ef7de5
2. Follow the build steps as mentioned in the README file
3. source setvars.sh
4. cd build
5. make config
6. make

**Expected behavior**
Successful build

**Code snips**
If applicable, add references to the software.

**System observed on:**
 - Hardware: PC-linux 
 - OS: [Ubuntu 18.04]
 - Versions [OSAL_5.3]

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Waheed Ejaz

jphickey, 2020-01-15:
Unfortunately, the README file is out of date.  We do need to fix this.  The classic build is being deprecated in favor of a CMake build.  See #339.

I'd like to understand your use case though.  Most users use OSAL in conjunction with CFE/CFS and in this case the build is integrated (OSAL is a component of the CFE build) and therefore there is generally no need to build OSAL by itself.  

Note that OSAL has no dependency on CFE, but there are some considerations that one needs to make when building it separately from whatever application is using it.  So the instructions for the README should include these aspects, hence why it is important to understand the proposed use-case.

waheedejaz, 2020-01-15:
I have to test IEEE P802.15.13 draft (LIFI MAC Layer) standard, as my master project. It is planned to be developed on linux using OSAL and ported to an embedded OS afterwards when the PHY layer is ready. OSAL really speeds up the development process and I can also do module level testing before integration to PHY. I also need to integrate OSAL to a existing project, just wanted to check if it works.
Just did a successful CMake build.

jphickey, 2020-01-15:
I don't think we should close this, this is a bona-fide issue that the instructions in the README for how to build the software in a test environment are incorrect.

Renaming this to cover the fact that the README instructions are wrong.

waheedejaz, 2020-01-15:
May I suggest adding cmake-options for the macros that are necessary for build:

- OSAL_SYSTEM_OSTYPE
- OSAL_SYSTEM_BSPTYPE
- ENABLE_UNIT_TESTS
- OSAL_INCLUDEDIR

Currently, if I run cmake-gui, I get errors before I hit "generate". Obviously, becuase the macros are not defined.

Also,
Can you please add contributing guidlines to the project? Can I grab this issue?

skliper, 2020-01-15:
> Can you please add contributing guidlines to the project?
Great suggestion, the are being worked right now!

> Can I grab this issue?
Let's coordinate with @jphickey, some of these may be handled by rescoping.

skliper, 2020-01-15:
Related to https://github.com/nasa/osal/issues/346

jphickey, 2020-01-15:
The issue I've had in the past with turning these items into actual CMake options as opposed to cache variables is that it seemed to cause issues where OSAL is a subdirectory of a bigger build (i.e. as in CFE).  In that use case, which is the more common one, the values for these variables would be supplied by the parent project, and would not be passed in directly by the user.  But its also possible that there is a way to supply options for child projects and I'm just not aware of it.

skliper, 2020-04-01:
> Can you please add contributing guidlines to the project? Can I grab this issue?
Note @waheedejaz that @astrogeco is working the contribution guidelines.

Also there's related discussion on #384.

PR #524 (2020-06-24): Fix #520, Update license and README for Apache 2.0, also updated quickstart
**Describe the contribution**
Update license and README for Apache 2.0 approval
Fix #520 
Fix #351 

**Testing performed**
CI (comment/document changes only)

**Expected behavior changes**
None

**System(s) tested on**
CI

**Additional context**
None

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GSFC
PR #521 (2020-06-25): Integration candidate: 2020-06-17
**Describe the contribution**
Fix #499
Fix #501
Fix #505 
Fix #506
Fix #520
Fix #351

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/106/checks

**Expected behavior changes**
PR #500 - Command line options in Linux are recognized, no longer ignored/dropped.

PR #504 - No impact to current unit testing which runs UT assert as a standalone app. Add a position independent code (PIC) variant of the ut_assert library, which can be dynamically loaded into other applications rather than running as a standalone OSAL application. This enables loadin [... cut]
closed 2020-06-25 by astrogeco