nasa/osal #363: "duplicate symbol '_main'" on macOS when building tests

stanislaw, 2020-02-27:
**Is your feature request related to a problem? Please describe.**

This is one of the issues addressed by #352. I am extracting it hereby to be considered separately from the rest of my work of macOS port.
 
I am getting linker errors on macOS because the test targets link in `libosal` library that has a `main()` function and they link `libut_bsp.a` which has a `main()` that launches tests. This causes a conflict with these test target' own main() function as follows:

```
duplicate symbol '_main' in:
    libosal.a(bsp_start.c.o)
    libut_bsp.a(bsp_ut.c.o)
```

**Describe the solution you'd like**

Usually the cases like this resolved by extracting the main function from one of the conflicting libraries. In this case it could make sense to extract the `main` function from `libosal` to a separate library such as `libosal-init` or `libosal-main` to avoid any conflicts with test targets. That separate library would then not be linked to the tests and the conflict would go away.

**Describe alternatives you've considered**

There seems to be no alternative to this behavior on macOS. You either have one `main()` function or there is a conflict.

**Additional context**

This is a more complete fragment of the output:

```
[9/17] Linking C executable timer-test
FAILED: timer-test
: && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -D_XOPEN_SOURCE=600    -D_LINUX_OS_ -Wl,-search_paths_first -Wl,-headerpad_max_install_names -u _OS_VolumeTable -u _OS_Application_Startup tests/CMakeFiles/timer-test.dir/timer-test/timer-test.c.o  -o timer-test  libut_assert.a libosal.a libut_bsp.a -lpthread -ldl posix-mac-addons/src/mqueue/libmqueue.a posix-mac-addons/src/stubs/librt.a posix-mac-addons/src/pthread/libposix-mac-pthread.a posix-mac-addons/src/semaphore/libposix-mac-semaphore.a posix-mac-addons/src/stubs/libposix-mac-stubs.a posix-mac-addons/src/time/libposix-mac-time.a posix-mac-addons/src/timer/libposix-mac-timer.a && :
duplicate symbol '_main' in:
    libosal.a(bsp_start.c.o)
    libut_bsp.a(bsp_ut.c.o)
ld: 1 duplicate symbol for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[10/17] Linking C executable bin-sem-test
FAILED: bin-sem-test
: && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -D_XOPEN_SOURCE=600    -D_LINUX_OS_ -Wl,-search_paths_first -Wl,-headerpad_max_install_names -u _OS_VolumeTable -u _OS_Application_Startup tests/CMakeFiles/bin-sem-test.dir/bin-sem-test/bin-sem-test.c.o  -o bin-sem-test  libut_assert.a libosal.a libut_bsp.a -lpthread -ldl posix-mac-addons/src/mqueue/libmqueue.a posix-mac-addons/src/stubs/librt.a posix-mac-addons/src/pthread/libposix-mac-pthread.a posix-mac-addons/src/semaphore/libposix-mac-semaphore.a posix-mac-addons/src/stubs/libposix-mac-stubs.a posix-mac-addons/src/time/libposix-mac-time.a posix-mac-addons/src/timer/libposix-mac-timer.a && :
duplicate symbol '_main' in:
    libosal.a(bsp_start.c.o)
    libut_bsp.a(bsp_ut.c.o)
ld: 1 duplicate symbol for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

**Requester Info**

Stanislav Pankevich

skliper, 2020-03-26:
@jphickey are your BSP updates addressing this?

jphickey, 2020-03-26:
> @jphickey are your BSP updates addressing this?

There are independent changes to combine BSP/PSP and simplify the link process (#312, #261) however I suspect that this particular issue has more to do with the patches for supporting MacOS.

skliper, 2020-04-01:
See https://github.com/nasa/osal/issues/384#issuecomment-607398611, may be related and has a short-term potential fix.

skliper, 2020-04-09:
@jphickey your clean-up likely fixes this?  If so self-assign?

skliper, 2020-04-13:
Likely solved by #404, @stanislaw could you confirm if it resolves your issue?

stanislaw, 2020-04-15:
Thanks for getting back to this. 

#404 is a big changeset so I am not sure if I can follow the exact vector of change but did it remove either of the main functions from the linkage?

I have rebased my branch in #352 and I can confirm that I see the issue like is described in the description of this issue:

```
duplicate symbol '_main' in:
    libosal.a(bsp_start.c.o)
    libut_bsp.a(bsp_ut.c.o)
ld: 1 duplicate symbol for architecture x86_64
```

I have double-check the changes in my branch and I see nothing in my changes that would additionally contribute to this issue.

This is what my commit is now built on top of:

```
commit 544f1d89f645ed8f2244a31f5c8346ae42166e79 (HEAD -> posix-mac-20200120-after-rebase-with-master, stanislaw/posix-mac-20200120-after-rebase-with-master)
Author: Stanislav Pankevich <s.pankevich@gmail.com>
Date:   Tue Dec 31 18:24:52 2019 +0100

    os/posix: port of the posix implementation to macOS

commit 7d9c4c8a470d3906050c5cf93203c4a9a9c09089 (origin/master, origin/HEAD)
Merge: 3024f63 21ea679
Author: Gerardo E. Cruz-Ortiz <59618057+astrogeco@users.noreply.github.com>
Date:   Thu Apr 9 19:21:04 2020 -0400

    Merge pull request #408 from nasa/integration-candidate

    Integration Candidate  2020-04-01
```

Hope this helps.

jphickey, 2020-04-15:
In particular the changes that are relevant here are for issues #261 and #312.

In your case, it looks like you were linking with both the OSAL BSP (bsp_start.c) as well as the UT BSP (bsp_ut.c).  In short, there was a thing (call it a hack) in the OSAL build that caused it to NOT build the OSAL BSP if you were building with CFE (because the PSP duplicated this), but that hack doesn't get invoked if you build OSAL standalone, which became apparent if unit tests were enabled because now you have a duplicate symbol.

This is all reworked in the pull #404.

stanislaw, 2020-04-15:
@jphickey it's been a while since I last time worked on my macos branch here so I didn't realize you are still integrating those changes in #404 and therefore they are not merged yet into the master branch. I have just tried to rebase my branch on top of your `jphickey/fix-312-261-362-build-cleanup` and I got some non-trivial merge conflicts. I will try it again tomorrow and let you know if it works on my side.

skliper, 2020-04-15:
I did just confirm on linux the issue was solved.  It showed up on there if CFE_SYSTEM_PSPNAME wasn't defined: `-DCFE_SYSTEM_PSPNAME=none` which is no longer necessary.  See #403 for the build implemented in CI (https://github.com/nasa/osal/blob/6967079c1591b51fcf277422c0b5d9880d7e29c6/.travis.yml)

stanislaw, 2020-04-16:
I have rebased my macOS branch on top of the `jphickey/fix-312-261-362-build-cleanup`. I can confirm that I can now built the tests without commenting out the `main()` function from the `bsp_start.c` file.

It is great to hear that #403 is being prepared. When both #403 and #404 are merged I would be happy to rebase my branch again and confirm this issue has gone on macOS.

skliper, 2020-06-05:
Closed by #404

PR #404 (2020-04-16): Fix #261, #312, and #362, OSAL build cleanup (multiple issues)
**Describe the contribution**

General cleanup and rework of the OSAL build scripts.   Fixes several issues:

Fixes #312 
Fixes #261 
Fixes #362 
Fixes #363 

These are combined into a single pull request as they each depend on the previous fix - they cannot be easily isolated into separate pull requests.  However, each specific issue fix is still an individual commit for review.

Summary of changes:
- Do not clobber (or generally even rely upon) the `CMAKE_C_FLAGS` variable, instead use the preferred approach of using the `target_ [... cut]
closed 2020-06-05 by skliper