nasa/osal #366: Remove FPU and exception handling APIs from OSAL

skliper, 2020-03-04:
**Is your feature request related to a problem? Please describe.**
Exception and FPU exception handling only partially implemented in OSAL

**Describe the solution you'd like**
Remove from OSAL, better suited to PSP if/when fully implemented.

**Describe alternatives you've considered**
None

**Additional context**
See also #18 to remove shared memory APIs

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-03-23:
Note FPUExecAttachHandler, FPUExcEnable and FPUExcDisable also don't have functional tests.  Certification issue

astrogeco, 2020-03-25:
CCB 2020-03-25: No objections

jphickey, 2020-04-21:
Note that these functions are being called by `src/unit-tests/oscore-test/ut_oscore_exception_test.c`.  As part of PR #427 I was testing by removing the FPU functions and got a link failure about this.  Will need to reconcile somehow.

I thought this was CCB approved but it looks like it hasn't been merged to master yet.  Is this waiting on anything?

skliper, 2020-04-21:
@jphickey that test was removed in #415?  @astrogeco hasn't merged yet.

jphickey, 2020-04-22:
That may explain it - I thought it was merged already.  Nonetheless in order for PR #427 to work with `OMIT_DEPRECATED` set then this is needed, so will need to rebase after this is merged.

PR #415 (2020-04-23): Fix #366, Deprecated exception handling APIs
**Describe the contribution**
Deprecates the FPU exception and exception handling APIs

Never fully implemented and more appropriate in PSP layer due to platform dependencies
Note - also removes old interrupt unit test header (missed in interrupt deprecation)
Fixes #366 

**Testing performed**
CI - https://travis-ci.com/github/skliper/cFS/builds/159870030

**Expected behavior changes**
None

**System(s) tested on**
CI - with master bundle

**Additional context**
May need slight mods relative to cmake related to listing c files vs aux

**Third party  [... cut]
commit 9ffab358 (2020-04-23): Merge pull request #415 from skliper/fix366-dep-fpu-except

Fix #366, Deprecated exception handling APIs
commit f45f565d (2020-04-23): Fix #366, Deprecated exception handling APIs

Never fully implemented and more appropriate
in PSP layer due to platform dependencies
commit b3571d6d (2020-05-01): Fix #366, Deprecated exception handling APIs

Never fully implemented and more appropriate
in PSP layer due to platform dependencies
closed 2020-04-27 by astrogeco