nasa/osal #367: OSAL fs defines scrub - missing prefix and unused

skliper, 2020-03-04:
**Is your feature request related to a problem? Please describe.**
FS_BASED, RAM_DISK, EEPROM_DISK, ATA_DISK, NUM_TABLE_ENTRIES all missing OS_ prefix
OS_CHK_ONLY and OS_REPAIR defined, but the input to the expected API is a boolean (bool repair)

**Describe the solution you'd like**
Add prefix, remove unused defines

**Describe alternatives you've considered**
None

**Additional context**
Do after #364 

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-04-10:
Started at https://github.com/skliper/osal/tree/fix367-fs-defs-scrub

Further work pending #404 merge to avoid conflicts

jphickey, 2020-04-23:
Recommend to deprecate `NUM_TABLE_ENTRIES` and replace with `OS_MAX_FILE_SYSTEMS`.  The latter already exists as part of #285 / PR #427 and the related scrub of osconfig.h parameters.

jphickey, 2020-05-14:
Proposing to deprecate the entire `OS_VolumeTable` concept and all symbols associated with this.  That is: `FS_BASED`, `RAM_DISK`, `EEPROM_DISK`, `ATA_DISK`, `NUM_TABLE_ENTRIES`.

Deprecating this also helps with #449.  We can keep the backward-compatibility layer in place for now, so the statically-defined `OS_VolumeTable[]` can still be linked in if `OMIT_DEPRECATED` is _not_ set.  But when it is set, this can be ignored completely.

PR #461 (2020-05-28): Fix #367, 453, 456, 460, Deprecate OS_VolumeTable and other fixes
**Describe the contribution**
Deprecate the OS_VolumeTable structure, which is no longer necessary with the current OSAL.

Remove the OS_VolumeTable definition from all BSPs, and provide a default (empty) one to support linking when OMIT_DEPRECATED is not set.

Note Due to unit test dependencies this PR also has to include fixes for other related issues.

Fix #367 
Fix #453 
Fix #456 
Fix #460

These are all in separate commits and they each build individually but there are some dependencies in making the unit tests work, so the [... cut]
commit 1c365692 (2020-05-28): Fix #367, Deprecate OS_VolumeTable objects

Remove the OS_VolumeTable definition from all BSPs, but provide
a default (empty) one to support linking when OMIT_DEPRECATED is
not set.
commit 922df4f1 (2020-05-28): Merge pull request #461 from jphickey/fix-367-deprecate-voltab

Fix #367, 453, 456, 460, Deprecate OS_VolumeTable and other fixes