nasa/osal #373: unit tests for OSAL network code

CDKnightNASA, 2020-03-19:
**Is your feature request related to a problem? Please describe.**
Seems there's not really any unit tests for the OSAL networking code.

**Describe the solution you'd like**
Should have unit tests.

**Describe alternatives you've considered**
Not having unit tests?

**Additional context**
Note that to do unit testing, it's likely we'll need to stub out OS-provided API's (open, select, socket, bind, accept, close().) Do we have a standard framework for stubbing out OS-provided functions?

**Requester Info**
Christopher.D.Knight@nasa.gov

CDKnightNASA, 2020-03-19:
No code yet, but would like to discuss at the CCB as to whether to stub out OS functions or otherwise how to test.

jphickey, 2020-03-19:
As noted in the CCB, there are two flavors of testing to consider:
1. Functional
2. Coverage

In OSAL the "tests" and "unit-tests" are functional tests (type 1).  These do _NOT_ use stubs., they use the real OSAL, linked to the real C library  We can (and should) write some new tests that exercise the network socket API.  We can create two threads, which talk to eachother via IP sockets that are bound to the localhost address (127.0.0.1).

OSAL also has coverage tests (type 2) but these are mainly to validate the error checking and other items which cannot typically be reached in the functional test.  This links each OSAL unit with stubs, including stubs for the C library calls.  Yes we have a framework for doing this, and the "shared" layer (including the network) should be covered by it already.  However the "impl" part probably needs better coverage, because only VxWorks is done 100% (that's a separate issue though).

skliper, 2020-03-20:
Concur, definitely need to add functional tests as Joe suggests.

skliper, 2020-03-23:
Missing tests for all of the following:
OS_SocketAddrInit, OS_SocketAddrToString, OS_SocketAddrFromString, OS_SocketAddrGetPort, OS_SocketAddrSetPort
OS_SocketOpen, OS_SocketBind, OS_SocketConnect, OS_SocketAccept, OS_SocketRecvFrom, OS_SocketSendTo, OS_SocketGetIdByName, OS_SocketGetInfo

Certification issue.

astrogeco, 2020-03-25:
CCB 2020-03-25: Will go ahead with this. It is a priority.

PR #549 (2020-08-18): fix #373, Add OSAL network APIs missing functional tests and fix #378 OS_TimedRead and OS_TimedWrite missing functional tests
fix #373, Add OSAL network APIs missing functional tests
fix #378, OS_TimedRead and OS_TimedWrite missing functional tests

**Describe the contribution**
Added OSAL network  and OS_TimedRead and OS_TimedWrite missing functional tests.

**Testing performed**
Steps taken to test the contribution:
Standard build and ran unit tests.

**System(s) tested on**
cFS Dev Server
OS: Ubuntu 18.04
Versions: OSAL 5.0.11.0

**Contributor Info** 
Yasir Majeed Khan
NASA GSFC/Emergent Sp [... cut]
commit 187b424f (2020-08-18): Fix #373, Add OSAL network APIs missing functional tests and fix #378 OS_TimedRead and OS_TimedWrite missing functional tests (#549)
PR #563 (2020-08-19): Integration Candidate: 2020-08-12
**Describe the contribution**
fix #373 
fix #378 
fix #550 
fix #379 

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/127/checks

**Expected behavior changes**
PR #549 - Adds OSAL network APIs missing functional tests as well as tests for OS_TimedRead and OS_TimedWrite 

PR #554 - Allows separate, dynamic registration of test setup and teardown routines which are executed before and after the normal test routine, which can create and delete any global/common test prerequisites.

PR #489 - Adds FileSysAddFixedMap missing functional API test
 [... cut]
closed 2020-08-19 by astrogeco