nasa/osal #374: Object Utility API's missing functional test

skliper, 2020-03-23:
**Is your feature request related to a problem? Please describe.**
OS_IdentifyObject, OS_ConvertToArrayIndex and OS_ForEachObject missing explicit functional tests

**Describe the solution you'd like**
Add tests

**Describe alternatives you've considered**
None.

**Additional context**
Certification violation

**Requester Info**
Jacob Hageman - NASA/GSFC

yammajamma, 2020-04-21:
I can take this one.

skliper, 2020-04-21:
Recommend following the osal/src/tests model.  Concept is to demonstrate these functionally work as intended/specified in the API (black box).  You can find the API here: https://github.com/nasa/cFS/blob/gh-pages/OSAL_Users_Guide.pdf

PR #439 (2020-05-26): Fix #374, Add Object Utility API functional tests
Fix #374, Object Utility API's missing functional test

**Describe the contribution**
Created missing functional tests for the OS_IdentifyObject, OS_ConvertToArrayIndex and OS_ForEachObject functions.  

**Testing performed**
Steps taken to test the contribution:
1. Standard build and ran unit tests.

**Expected behavior changes**

**System(s) tested on**
cFS Dev Server
OS: Ubuntu 18.04
Versions: OSAL 5.0.11.0

**Contributor Info **
Yasir Khan
NASA GSFC
commit 35d8b731 (2020-05-26): Merge pull request #439 from yammajamma/fix374-2-Object-Utilities-Missing-Tests

Fix #374, Add Object Utility API functional tests
commit 74669973 (2020-05-26): Fix #374, Updated OS_ConvertToArrayIndex to verify output is in range between 0 and <MAX and output array indices vary
commit 8352d2fa (2020-05-26): Fix #374, Add idmap functional tests
closed 2020-05-29 by astrogeco