nasa/osal #376: OS_IntGetMask OS_IntSetMask missing functional tests

skliper, 2020-03-23:
**Is your feature request related to a problem? Please describe.**
Missing functional tests for OS_IntGetMask and OS_IntSetMask

**Describe the solution you'd like**
Add tests

**Describe alternatives you've considered**
None

**Additional context**
Certification issue

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2020-03-23:
Not so sure how these can be tested in a functional test as the definition of an interrupt mask is CPU/BSP-specific.  While OSAL might have a wrapper around the syscall it doesn't really abstract the semantics of it.  Furthermore, the side effects are likely quite different on a multi-core CPU.

Recommend to deprecate the functions, like the FPU variants.  I don't see either of these calls being used by any current CFE/PSP code.  POSIX and RTEMS both return `OS_ERR_NOT_IMPLEMENTED` for these anyway.

skliper, 2020-03-23:
@acudmore opinion?  I'm all for deprecation.

skliper, 2020-03-25:
#71 also related (and highlights issue of platform dependence).

skliper, 2020-03-26:
#396 makes this no longer applicable

closed 2020-03-26 by skliper