nasa/osal #377: OSAL Select APIs missing functional test

skliper, 2020-03-23:
**Is your feature request related to a problem? Please describe.**
OS_SelectMultipe, OS_SelectSingle, OS_SelectFdZero, OS_SelectFdAdd, OS_SelectFdClear, OS_SelectFdIsSet all missing functional tests

**Describe the solution you'd like**
Add tests

**Describe alternatives you've considered**
None

**Additional context**
Certification issue.

**Requester Info**
Jacob Hageman - NASA/GSFC

CDKnightNASA, 2020-03-26:
I'll start taking a crack at this.

CDKnightNASA, 2020-04-03:
will wait for #405 to be merged into master and will use new macros, plus need to rebase and remerge the code from #391 and #392

skliper, 2020-05-13:
This was closed by mistake, Select APIs still need a test to show they function correctly

jphickey, 2020-05-13:
FYI - there _IS_ now some unit test code which calls "select" .... 

I have to make a change in the select API anyway to fix a different bug (#453), and change I'm making for that issue also requires some updates to these test cases.  This is actually how the bug was noticed -- running this test on RTEMS caused failure.

skliper, 2020-08-11:
TODO - check if #462 is sufficient to satisfy the unit test standards.

commit e7b58e77 (2020-04-21): fix #377, fix #390, fix #392 - osal select API unit tests and fixes
commit e4b37898 (2020-05-01): fix #377, fix #390, fix #392 - osal select API unit tests and fixes
PR #683 (2020-12-18): Fix #377, Add "OSAL Select" functional tests
**Describe the contribution**
Fixes #377 
Added new functional tests for the the select and select multiple functions. 

**Testing performed**
Ran the new tests. 

**Expected behavior changes**
None

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Alex Campbell GSFC
commit 37059349 (2020-12-18): Merge pull request #683 from zanzaben/fix377_Select_Functional_Tests

Fix #377, Add "OSAL Select" functional tests
closed 2020-12-18 by astrogeco