nasa/osal #379: OS_FileSysAddFixedMap missing functional test

skliper, 2020-03-23:
**Is your feature request related to a problem? Please describe.**
Missing functional test for OS_FileSysAddFixedMap

**Describe the solution you'd like**
Add test

**Describe alternatives you've considered**
None

**Additional context**
Certification issue

**Requester Info**
Jacob Hageman - NASA/GSFC

yammajamma, 2020-05-14:
I can take this one.

jphickey, 2020-05-14:
FYI - as part of my upcoming changes (#449 #460), the unit tests _will_ be using this function to set up for a BSP-agnostic file map.  

I'm not sure if there is anything more that can be done with them beyond this, because any test would require going around the OSAL abstraction in a platform-dependent manner.  Furthermore, if they didn't work right, all the tests following that depend on the map will fail.

So in other words I think its covered already

skliper, 2020-05-27:
@yammajamma - could you review the unit test updates in #461 and ensure it meets the functional test standards (covers everything we've discussed a functional test needs to cover)?  If so we can close this.  If not let's add whatever is missing and/or create a unique test.

yammajamma, 2020-05-27:
@skliper Yes, I can take a look at that, thanks.

PR #489 (2020-08-14): Fix #379, Add FileSysAddFixedMap functional API test
Fix #379, Add FileSysAddFixedMap functional API test.

**Describe the contribution**
Added FileSysAddFixedMap functional API test

**Testing performed**
Steps taken to test the contribution:
1. Standard build and ran unit tests.

**System(s) tested on**
cFS Dev Server
OS: Ubuntu 18.04
Versions: OSAL 5.0.11.0

**Contributor Info - All information REQUIRED for consideration of pull request**
Yasir Majeed Khan
NASA GSFC/Emergent Space
commit 34e09780 (2020-08-14): Fix #379, Add FileSysAddFixedMap functional API test (#489)
PR #563 (2020-08-19): Integration Candidate: 2020-08-12
**Describe the contribution**
fix #373 
fix #378 
fix #550 
fix #379 

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/127/checks

**Expected behavior changes**
PR #549 - Adds OSAL network APIs missing functional tests as well as tests for OS_TimedRead and OS_TimedWrite 

PR #554 - Allows separate, dynamic registration of test setup and teardown routines which are executed before and after the normal test routine, which can create and delete any global/common test prerequisites.

PR #489 - Adds FileSysAddFixedMap missing functional API test
 [... cut]
closed 2020-08-19 by astrogeco