nasa/osal #380: TimeBase API missing functional tests

skliper, 2020-03-23:
**Is your feature request related to a problem? Please describe.**
OS_TimeBaseCreate, OS_TimeBaseSet, OS_TimeBaseDelete, OS_TimeBaseGetIdByName, OS_TimeBaseGetInfo, OS_TimeBaseGetFreeRun all missing functional tests

**Describe the solution you'd like**
Add tests

**Describe alternatives you've considered**
None

**Additional context**
Certification Issue

**Requester Info**
Jacob Hageman - NASA/GSFC

yammajamma, 2020-05-14:
I can take this.

jphickey, 2020-05-14:
Note that a time base object exists primarily to underpin the timer objects, and they aren't used directly by apps.  Therefore API is used and invoked, either directly or indirectly, as part of the existing timer tests.  I'm also not sure how you could even test it in isolation, because these objects aren't used directly.  Aside from simply creating/deleting the objIDs and just confirming this works?

skliper, 2020-05-14:
It's basically to meet the formal test requirements for certification, test full parameter ranges, etc.

jphickey, 2020-05-14:
Yeah, most of that parameter checking is already in the coverage test, but could be done again in a full-stack/non-stub enviornment.  

I guess the other thing is to confirm you can create e.g.  up to `OS_MAX_TIMEBASES` objects and the +1 one fails, and also that you can delete them successfully too?

skliper, 2020-05-14:
Yes please (create all, delete all)!  Note we don't get to take verification credit from a coverage test.  So the parameter checking needs to be done in the functional test.

PR #481 (2020-06-08): Fix #380, Add OS_TimeBase Api Functional Tests
Fix #380, Add OS_TimeBase Api Functional Tests

**Describe the contribution**
Added OS_TimeBase Api Functional Tests for OS_TimeBaseCreate, OS_TimeBaseSet, OS_TimeBaseDelete, OS_TimeBaseGetIdByName, OS_TimeBaseGetInfo, OS_TimeBaseGetFreeRun

**Testing performed**
Steps taken to test the contribution:
1. Standard build and ran unit tests.

**System(s) tested on**
cFS Dev Server
OS: Ubuntu 18.04
Versions: OSAL 5.0.11.0

**Contributor Info - All information REQUIRED for consideration of pull request**
Yasir Khan
NASA GSFC
commit 47317552 (2020-06-08): Fix #380, Add OS_TimeBase Api Functional Tests
commit f3f8b8a7 (2020-06-08): fix #380, Updated test to implement changes requested
commit ff44e9f9 (2020-06-08): Merge pull request #481 from yammajamma/fix380-Add-TimeBase-Api-Functional-Tests

Fix #380, Add OS_TimeBase Api Functional Tests
closed 2020-06-10 by astrogeco