nasa/osal #424: Test  #7: osal-core-test Failure

yammajamma, 2020-04-17:
**Describe the bug**
Test  #7: osal-core-test fails.  

**To Reproduce**
Steps to reproduce the behavior:
Note: this test was not readily repeatable, but it failed 2/10 times on this branch: 
fix413-Add-Reference-To-Osal-Users-Guide-From-ReadMe
1. Go to branch stated above
2. Run build/exe/cpu1/osal-core-test

**Expected behavior**
The test failed 2/10 times.  Specifically, failure is due to the code snippet noted below in  osal-core-test.c, line 138 where a small time delay is supposed to occur and the child task is perhaps not exiting before the OS_TaskDelete call happens, so it's returning success. 

**Code snips**
[ FAIL] 01.261 osal-core-test.c:138 - OS_TaskDelete, self exiting task

**System observed on:**
GNOME VM
OS: Ubuntu 18.04
Versions: OSAL 5.0.11.0

**Additional context**
This may be due to the VM delay.  

**Reporter Info**
Yasir Khan
NASA GSFC

skliper, 2020-04-17:
https://github.com/nasa/osal/blob/7d9c4c8a470d3906050c5cf93203c4a9a9c09089/src/tests/osal-core-test/osal-core-test.c#L121-L140

stanislaw, 2020-04-17:
It is great to see this reported by someone else on Linux. My macOS port has this too this is why I have commented out the tests in the branch. See [here](https://github.com/nasa/osal/pull/352/files#diff-3f5cde84737720036ad3a3e28edd89d9R138).

skliper, 2020-04-20:
Perhaps loop on the delay checking if the child task has exited.

PR #426 (2020-04-28): Fix #424, Add looping wait for self exiting task
**Describe the contribution**
Loops the parent OS_TaskDelay until child task exits before testing OS_TaskDelete.  Avoids the test depending on system loading/performance with single delay approach.
Fix #424 

**Testing performed**
Steps taken to test the contribution:
1. Standard build, ran `osal-core-test -d` and observed Loop count of 1 (I've never experienced the failure, which if avoided would have a loop count > 1)

**Expected behavior changes**
Consistent osal-core-test execution status

**System(s) tested on**
 - Hardware: cFS Dev Server  [... cut]
commit 3707b788 (2020-04-28): Merge pull request #426 from skliper/fix424-self-exit-task-delay

Fix #424, Add looping wait for self exiting task
commit e7bba8f0 (2020-04-28): Fix #424, Add looping wait for self exiting task
commit 36251778 (2020-05-01): Fix #424, Add looping wait for self exiting task
closed 2020-05-01 by astrogeco