nasa/osal #425: Test_OS_ConvertToArrayIndex: tautological assertion

lethuillierg, 2020-04-18:
**Describe the bug**
Unit test ` Test_OS_ConvertToArrayIndex` implements the following assertion: 
https://github.com/nasa/osal/blob/7d9c4c8a470d3906050c5cf93203c4a9a9c09089/src/unit-test-coverage/shared/src/coveragetest-idmap.c#L610-L611

This is a tautological assertion as it compares `local_idx1` with itself.

**To Reproduce**
N/A (code analysis)

**Expected behavior**
`local_idx1` should be compared with `local_idx2`:
```
UtAssert_True(local_idx1 == local_idx2, "local_idx1 (%lu) == local_idx2 (%lu)",
            (unsigned long)local_idx1, (unsigned long)local_idx2);
```

**Code snips**
See above.

**System observed on:**
N/A (code analysis)

**Additional context**
N/A

**Reporter Info**
Guillaume Lethuillier

skliper, 2020-04-20:
@yammajamma - want to take this one?

yammajamma, 2020-04-20:
@skliper Yes!

PR #428 (2020-04-28): Fix #425, Correct Test_OS_ConvertToArrayIndex assertion typo
Fix #425 Test_OS_ConvertToArrayIndex: tautological assertion

**Describe the contribution**
Unit test Test_OS_ConvertToArrayIndex implements the following assertion:
osal/src/unit-test-coverage/shared/src/coveragetest-idmap.c

Lines 610 to 611 in 7d9c4c8

 UtAssert_True(local_idx1 == local_idx1, "local_idx1 (%lu) == local_idx2 (%lu)", 
         (unsigned long)local_idx1, (unsigned long)local_idx2); 

This is a tautological assertion as it compares local_idx1 with itself.  

Update:
local_idx1 is now compared with local_idx2:

UtAsser [... cut]
commit eb0230f2 (2020-04-28): Fix #425, Correct Test_OS_ConverToArrayIndex assertion typo
commit f31e2d94 (2020-04-28): Merge pull request #428 from yammajamma/fix-425-Tautological-Assertion-Fix

Fix #425, Correct Test_OS_ConvertToArrayIndex assertion typo
commit e0c8fbf3 (2020-05-01): Fix #425, Correct Test_OS_ConverToArrayIndex assertion typo
closed 2020-05-01 by astrogeco