nasa/osal #435: Unhandled output truncation errors when building on CentOS 8.1

lethuillierg, 2020-04-22:
**Describe the bug**
On CentOS 8.1, when cFS is being built with enabled unit tests, `make` returns the following unhandled output truncation errors (`-Werror=format-truncation=`) related to osal unit tests:

```
[cfs@localhost cFS]$ make
make --no-print-directory -C "build" mission-all
[100%] Built target elf2cfetbl
[100%] Built target mission-version
[100%] Built target mission-prebuild
[  1%] Built target osal_posix_impl
[  2%] Built target osal_pc-linux_impl
[  3%] Built target osal_bsp
[  9%] Built target osal

[. . . omitting lines . . .]

[ 65%] Building C object osal/unit-tests/osfile-test/CMakeFiles/osal_file_UT.dir/ut_osfile_dirio_test.c.o
In file included from /home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.h:14,
                 from /home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c:11:
/home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c: In function ‘UT_os_makedir_test’:
/home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c:157:31: error: ‘%s’ directive output may be truncated writing up to 73 bytes into a region of size 30 [-Werror=format-truncation=]
     UT_os_sprintf(g_fileName, "%s/mkdir_File.txt", g_dirName);
                               ^~~~~~~~~~~~~~~~~~~  ~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/inc/ut_os_support.h:82:30: note: in definition of macro ‘UT_os_sprintf’
     snprintf(buf,sizeof(buf),__VA_ARGS__)
                              ^~~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/inc/ut_os_support.h:82:5: note: ‘snprintf’ output between 16 and 89 bytes into a destination of size 30
     snprintf(buf,sizeof(buf),__VA_ARGS__)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c:157:5: note: in expansion of macro ‘UT_os_sprintf’
     UT_os_sprintf(g_fileName, "%s/mkdir_File.txt", g_dirName);
     ^~~~~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c: In function ‘UT_os_readdir_test’:
/home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c:465:37: error: ‘snprintf’ output may be truncated before the last format character [-Werror=format-truncation=]
     UT_os_sprintf(g_subdirNames[0], "%s/%s", g_dirName, g_tgtSubdirs[0]);
                                     ^~~~~~~
/home/cfs/cFS/osal/src/unit-tests/inc/ut_os_support.h:82:30: note: in definition of macro ‘UT_os_sprintf’
     snprintf(buf,sizeof(buf),__VA_ARGS__)
                              ^~~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/inc/ut_os_support.h:82:5: note: ‘snprintf’ output 2 or more bytes (assuming 75) into a destination of size 74
     snprintf(buf,sizeof(buf),__VA_ARGS__)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c:465:5: note: in expansion of macro ‘UT_os_sprintf’
     UT_os_sprintf(g_subdirNames[0], "%s/%s", g_dirName, g_tgtSubdirs[0]);
     ^~~~~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c:475:37: error: ‘snprintf’ output may be truncated before the last format character [-Werror=format-truncation=]
     UT_os_sprintf(g_subdirNames[1], "%s/%s", g_dirName, g_tgtSubdirs[1]);
                                     ^~~~~~~
/home/cfs/cFS/osal/src/unit-tests/inc/ut_os_support.h:82:30: note: in definition of macro ‘UT_os_sprintf’
     snprintf(buf,sizeof(buf),__VA_ARGS__)
                              ^~~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/inc/ut_os_support.h:82:5: note: ‘snprintf’ output 2 or more bytes (assuming 75) into a destination of size 74
     snprintf(buf,sizeof(buf),__VA_ARGS__)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c:475:5: note: in expansion of macro ‘UT_os_sprintf’
     UT_os_sprintf(g_subdirNames[1], "%s/%s", g_dirName, g_tgtSubdirs[1]);
     ^~~~~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c: In function ‘UT_os_rewinddir_test’:
/home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c:587:37: error: ‘snprintf’ output may be truncated before the last format character [-Werror=format-truncation=]
     UT_os_sprintf(g_subdirNames[0], "%s/%s", g_dirName, g_tgtSubdirs[0]);
                                     ^~~~~~~
/home/cfs/cFS/osal/src/unit-tests/inc/ut_os_support.h:82:30: note: in definition of macro ‘UT_os_sprintf’
     snprintf(buf,sizeof(buf),__VA_ARGS__)
                              ^~~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/inc/ut_os_support.h:82:5: note: ‘snprintf’ output 2 or more bytes (assuming 75) into a destination of size 74
     snprintf(buf,sizeof(buf),__VA_ARGS__)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/cfs/cFS/osal/src/unit-tests/osfile-test/ut_osfile_dirio_test.c:587:5: note: in expansion of macro ‘UT_os_sprintf’
     UT_os_sprintf(g_subdirNames[0], "%s/%s", g_dirName, g_tgtSubdirs[0]);
     ^~~~~~~~~~~~~
[... 60 lines cut ...]

**Expected behavior**
`make` should not return any error.

**Code snips**
N/A

**System observed on:**
 - iMac (Retina 4K, 2019)
 - OS: CentOS 8.1 (`centos-release-8.1-1.1911.0.8.el8.x86_64`) (via VMware Fusion 11.5.3) / minimal installation / packages installed: git, make, cmake, gcc.
 - GCC: `8.3.1 20190507 (Red Hat 8.3.1-4)`
 - Versions: master bundle

**Additional context**
For convenience and mainly to investigate the possibility of performing automated end-to-end testing (for instance, to make a pseudo-GSW deterministically interact with the FSW and check the telemetry), I have taken the initiative to dockerize cFS. 

In this context, these errors occurred when creating a CentOS-based Docker image. Their occurrence was then confirmed using a CentOS virtual machine (the logs provided above come from the unit tests performed on the virtual machine).

No such errors occur on (a virtualized) Ubuntu 18.04. This behavior seems then specific to CentOS, or perhaps to a certain version of GCC.

(Because these errors are directly related to osal, I have decided to open this issue here instead of the cFS repository).

**Reporter Info**
Guillaume Lethuillier
Personal

jphickey, 2020-04-22:
GCC has been adding more and more validation/verification in recent releases, so these warnings are almost certainly related to the newer GCC version here.

Worth noting that @CDKnightNASA is actively working on putting these scattered string manipulations for error reporting into common code (e.g. nasa/cfe#492) which may either fix these or make them easier to fix once this effort is done.

lethuillierg, 2020-04-22:
Indeed: I have just installed GCC 8.4.0 (`gcc (Ubuntu 8.4.0-1ubuntu1~18.04) 8.4.0`) on Ubuntu and now the same errors occur.

jphickey, 2020-04-22:
Thanks for the confirmation... in the meantime you might try `-Wno-format` (IIRC) to avoid the warning as an interim workaround.  You can add to `sample_defs/global_build_options.cmake`.  Either that or remove `-Werror`.  I'd rather not try to fix these until the other UT refactoring work is done, as it will likely be impossible to merge.

lethuillierg, 2020-04-22:
Thank you @jphickey. 

At this time, I will directly install GCC 7 in my Docker images until cFS becomes fully compatible with GCC 8, but I keep your advice in mind.

> I'd rather not try to fix these until the other UT refactoring work is done, as it will likely be impossible to merge.

I agree.

jphickey, 2020-04-27:
See related: nasa/cfe#641.

I'll also be rolling some fixes in for this with #285, as doing it separately would likely create a merge conflict.

PR #441 (2020-05-11): Fix #435, fix string manipulations to avoid warnings
**Describe the contribution**
Resize buffers and add explicit termination to string copies.
Avoids warnings on GCC9 with strict settings and optimization enabled.

Fixes #435

**Testing performed**
Build with default config and gcc 9.3.0 and confirm warnings are gone

**Expected behavior changes**
Warnings fixed.

**System(s) tested on**
Ubuntu 20.04 LTS

**Additional context**
Currently a draft because it uses PR #427 as a baseline.  Will mark as non-draft once prerequisite is merged.

**Contributor Info - All information REQUIRED for consi [... cut]
commit a227f3b9 (2020-05-11): Fix #435, Changes for string ops warnings

Resize buffers and tweak string copies to avoid truncation warnings
in GCC 9.  Note the code was generally OK (handled truncation) but
the new compiler still generates warnings even with correct usage
of certain functions.

In the UT_os_printf macro, be sure to "use" the return value
of the snprintf() call, which avoids warnings about truncation.
commit c1555b81 (2020-05-11): Merge pull request #441 from jphickey/fix-435-string-ops

Fix #435, fix string manipulations to avoid warnings
PR #458 (2020-05-13): Integration Candidate COMBINED 2020-04-29 and 2020-05-06
**Describe the contribution**

Fix #437, add new API for obtaining exception task ID
Fix #435, Changes for string ops warnings 

**Testing performed**
See PRs
Bundle CI - https://travis-ci.com/github/nasa/cFS/builds/165726730

**Expected behavior changes**
PR #441 - Resized buffers and added explicit termination to string copies. No warnings on GCC9 with strict settings and optimization enabled.

PR #446 - New API to reverse lookup an OS-provided thread/task identifier back to an OSAL ID.
Any use of existing OStask_id field within the ta [... cut]
closed 2020-05-13 by astrogeco