nasa/osal #437: Deprecate OStask_id field from OS_task_prop_t

jphickey, 2020-04-23:
**Is your feature request related to a problem? Please describe.**
This field is not necessarily applicable to all OS types and it breaks the abstraction.  OSAL should not be reporting the raw/unabstracted values back to the application.

This field is fundamentally broken in any environment where the underlying OS task ID isn't convertible to a `uint32`.  This includes real systems, such as:

- Cygwin, where `pthread_t` is a compound data type, not an integer (POSIX specifically allows this)
- Systems where the task ID is actually pointer to the TCB (in which case it will be 64 bits on 64 bit systems)

Currently this field serves one purpose, which is to allow the CFE ES exception processing to find the OSAL task ID associated with the exception.   However, this design is being reworked in nasa/cfe#411 where more of the exception processing is done in the CFE PSP.  With this, there should be no need for applications to ever use the `OStask_id` member.

**Describe the solution you'd like**
The `OStask_id` member of `OS_task_prop_t` should be marked as deprecated.  As a replacement, an new API can be added which can be invoked by the PSP/BSP to aid in exception processing.  This new API should use an abstract pointer and size, not assume that the task identifier information fits within a 32-bit integer.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2020-04-23:
@skliper, @acudmore, @jwilmot - please comment if you have any thoughts on this, I'd like to include this as part of the overall strategy for fixing nasa/cfe#411 and nasa/cfe#76.

skliper, 2020-04-23:
Concur.

acudmore, 2020-04-23:
The proposed solution sounds good to me.

PR #446 (2020-05-08): Fix #437, Add new API for obtaining exception task ID
**Describe the contribution**
The OS_task_prop_t had contained an `OStask_id` field which was used for reverse-mapping the OSAL task ID during exception processing by CFE.  Unfortunately this was a uint32 and not compatible with any OS which used task identifiers that were not convertible to such a value (e.g. `pthread_t`, or 64-bit pointers, etc).

Deprecate the old field and replace with a new API that performs reverse lookup within OSAL using an abstract pointer and size, so its compatible with any OS task representation object.

Fix #43 [... cut]
commit 998377b9 (2020-05-08): Merge pull request #446 from jphickey/fix-437-exception-api

Fix #437, Add new API for obtaining exception task ID
commit f3f3b65c (2020-05-08): Fix #437, add new API for obtaining exception task ID

Deprecate the OStask_ID field within the property structure.

Replace with a new API to perform a task ID reverse-lookup
based on an abstract data object (the system task ID).
PR #458 (2020-05-13): Integration Candidate COMBINED 2020-04-29 and 2020-05-06
**Describe the contribution**

Fix #437, add new API for obtaining exception task ID
Fix #435, Changes for string ops warnings 

**Testing performed**
See PRs
Bundle CI - https://travis-ci.com/github/nasa/cFS/builds/165726730

**Expected behavior changes**
PR #441 - Resized buffers and added explicit termination to string copies. No warnings on GCC9 with strict settings and optimization enabled.

PR #446 - New API to reverse lookup an OS-provided thread/task identifier back to an OSAL ID.
Any use of existing OStask_id field within the ta [... cut]
closed 2020-05-13 by astrogeco