nasa/osal #438: #include <sys/signal.h> from os-posix.h causes compiler warnings in Alpine Linux

astrogeco, 2020-04-28:
**Describe the bug**
When trying to build the cFS bundle using Alpine Linux using the musl library we get a gcc error:
`#warning redirecting incorrect #include <sys/signal.h> to <signal.h>`

From my seaarches It seems that `sys/signal.h`  is an older implementation.

**To Reproduce**
Build and run the following Docker image
```
# Dockerfile to make cFS testing platform independent. Based on .travis.yml from cfs-Bundle repo
FROM alpine:3.11

RUN apk add \
  build-base \
  cmake \
  git

RUN echo -e "http://dl-cdn.alpinelinux.org/alpine/edge/community\nhttp://dl-cdn.alpinelinux.org/alpine/edge/main" >> /etc/apk/repositories
RUN apk add --no-cache -X http://dl-cdn.alpinelinux.org/alpine/edge/testing \
    perf

# Force image rebuild if bundle has changed since last build
ADD https://api.github.com/repos/nasa/cfs/git/refs/heads/master version.json

# Get current bundle
RUN git clone --recurse-submodules -j8 https://github.com/nasa/cFS.git

# Define cfs environment variables
ENV \
  BUILDTYPE=debug \
  CFS_ROOT=/cFS \
  ENABLE_UNIT_TESTS=true \
  OMIT_DEPRECATED=false \
  SIMULATION=native

# Set up for build
RUN cp $CFS_ROOT/cfe/cmake/Makefile.sample $CFS_ROOT/Makefile
RUN cp -r $CFS_ROOT/cfe/cmake/sample_defs $CFS_ROOT/sample_defs

# Run make at startup
ENTRYPOINT make -C $CFS_ROOT

```

Build and run the container:

Build: `docker build --tag cfs-bundle:alpine DOCKERFILE_DIRECTORY
Run: `docker run -it --rm cfs-bundle:alpine`

**Expected behavior**
cFS Build like with Travis CI and Ubuntu

**Code snips**
In `os-posix.h:40`: `#include <sys/signal.h>`

**System observed on:**
Docker Desktop 2.2.0.4 
OSX 10.14.6

lethuillierg, 2020-04-28:
Currently working on this topic (https://github.com/nasa/cFS/issues/56#issuecomment-619593513), I came across this issue too.

I have observed the following behaviors after having merely replaced `#include <sys/signal.h>` with `#include <signal.h>` in `osal/src/os/posix/os-posix.h` (using the following command, before `make prep`: `RUN sed -i 's+#include <sys/signal.h>+#include <signal.h>+' osal/src/os/posix/os-posix.h`).

**Alpine 3.11 (Docker, GCC 9.2.0-r4, native, omit deprecated)**

**_Case 1: `BUILDTYPE=debug`_**

- Build: failure

```
Step 18/25 : RUN make

[. . . omitting lines . . .]

[ 25%] Linking C executable coverage-vxworks-osshell-testrunner
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: cannot find gcrt1.o: No such file or directory
collect2: error: ld returned 1 exit status
make[7]: *** [osal/unit-test-coverage/vxworks/CMakeFiles/coverage-vxworks-osshell-testrunner.dir/build.make:92: osal/unit-test-coverage/vxworks/coverage-vxworks-osshell-testrunner] Error 1
make[6]: *** [CMakeFiles/Makefile2:533: osal/unit-test-coverage/vxworks/CMakeFiles/coverage-vxworks-osshell-testrunner.dir/all] Error 2
make[5]: *** [Makefile:141: all] Error 2
make[4]: *** [CMakeFiles/native-all.dir/build.make:57: CMakeFiles/native-all] Error 2
make[3]: *** [CMakeFiles/Makefile2:358: CMakeFiles/native-all.dir/all] Error 2
make[2]: *** [CMakeFiles/Makefile2:311: CMakeFiles/mission-all.dir/rule] Error 2
make[1]: *** [Makefile:279: mission-all] Error 2
make: *** [Makefile:120: all] Error 2
```

**_Case 2: `BUILDTYPE=release`_**

- Build: failure _(seemingly related to the version of GCC installed by default on Alpine 3.11)_

```
Step 18/25 : RUN make

[. . . omitting lines . . .]

[ 50%] Building C object tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/elf2cfetbl.c.o
/cFS/tools/elf2cfetbl/elf2cfetbl.c: In function 'ProcessCmdLineOptions':
/cFS/tools/elf2cfetbl/elf2cfetbl.c:945:13: error: 'strncpy' specified bound 32 equals destination size [-Werror=stringop-truncation]
  945 |             strncpy(Description, &Arguments[i][2], 32);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/cFS/tools/elf2cfetbl/elf2cfetbl.c:939:13: error: 'strncpy' specified bound 38 equals destination size [-Werror=stringop-truncation]
  939 |             strncpy(TableName, &Arguments[i][2], 38);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[4]: *** [tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/build.make:63: tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/elf2cfetbl.c.o] Error 1
make[3]: *** [CMakeFiles/Makefile2:477: tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/all] Error 2
make[2]: *** [CMakeFiles/Makefile2:311: CMakeFiles/mission-all.dir/rule] Error 2
make[1]: *** [Makefile:279: mission-all] Error 2
make: *** [Makefile:120: all] Error 2
```

Additional work will then probably be required to make cFS successfully build on Alpine.

**CentOS 7 (Docker, GCC 4.8.5, native, debug, omit deprecated)**

- Build: OK
- Unit tests: failure—`osal_timer_UT` fails. (It does not fail when the original `#include <sys/signal.h>` is used instead of `#include <signal.h>`).

```
48/56 Testing: osal_timer_UT

[. . . omitting lines . . .]

[ FAIL] 05.004 ut_ostimer_timerio_test.c:576 - #4 Nominal
[  END] 05 OS_TimerSet          TOTAL::4     PASS::2     FAIL::1      MIR::0      TSF::0      N/A::0   

[  END] 05 SUMMARY              TOTAL::21    PASS::17    FAIL::1      MIR::0      TSF::0      N/A::0   
```
This failure, if confirmed, should be taken into consideration if it is decided to modify the include.

**Ubuntu 18.04 (Docker and VM, GCC 7.4.0, native, debug, omit deprecated)**
- Build: OK
- Unit tests: OK
- Runtime: OK

lethuillierg, 2020-05-03:
Complementary tests reveal that `osal_timer_UT` failure is likely related to https://github.com/nasa/osal/issues/335. Therefore, it is probably not specific to the current issue.

closed 2026-02-25 by dzbaker