nasa/osal #450: Add alternative source directory option for OSAL implementations

jphickey, 2020-05-08:
**Is your feature request related to a problem? Please describe.**
The OSAL build script always looks for BSP implementations in:
    `${OSAL_SOURCE_DIR}/src/bsp/${OSAL_SYSTEM_BSPTYPE}`
Likewise it always looks for OS implementations in:
    `${OSAL_SOURCE_DIR}/src/os/${OSAL_SYSTEM_OSTYPE}`

This presents a challenge to for a user with a custom OS/BSP implementation not within the mainline OSAL source tree, but wishes to use the github repo directly.

**Describe the solution you'd like**
It should be possible to provide an external directory that the OSAL build can use.

**Describe alternatives you've considered**
Create an OSAL fork or "git subtree" to assemble a full CFE/CFS repo, which allows one to add custom packages or patches they need.

**Additional context**
The internal API between the BSP/OS implementation components is not stabilized like the public API is, and can change any time.  

If directly using a "master" branch from github in conjunction with a locally-controlled implementation module, users may experience frequent breakage, as the component interface can change any time rendering it incompatible with their local version.  

In contrast, although the fork/subtree approach does have a similar issue, it doesn't happen "automatically" - it requires a specific user action to pull and merge the new "master" so it creates a buffer that avoids unsolicited breakage.

This feature should come with a big disclaimer that says its for debug/development only, not to use it in production.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2020-05-08:
Actually... is it possible now by defining `OSAL_SYSTEM_BSPTYPE` as something like `../../../myosalbsp`?  Not exactly sanctioned and likely not the original intent... but doesn't require any changes in OSAL?

skliper, 2020-05-08:
fragile obviously... but so is depending on internal APIs

PR #468 (2020-05-26): Fix #450, add external source directory for OS/BSP
**Describe the contribution**

If the "OSAL_EXT_SOURCE_DIR" cache variable is set, this location will be checked first for a BSP/OS implementation layer.  This can point to an out-of-tree implementation layer if necessary.

Fixes #450

**Testing performed**
Configure a test project to use a BSP provided in the external source dir, and confirm that it builds correctly and sanity check result (unit tests, CFE runs normally).

**Expected behavior changes**
No impact to behavior unless user requires a BSP/OS implementation which is not part of the [... cut]
commit 063d69f9 (2020-05-26): Merge pull request #468 from jphickey/fix-450-alt-source-directory

Fix #450, add external source directory for OS/BSP
commit c139ae38 (2020-05-26): Fix #450, add external source directory for OS/BSP

If the "OSAL_EXT_SOURCE_DIR" cache variable is set, this location
will be checked first for a BSP/OS implementation layer.  This can
point to an out-of-tree implementation layer if necessary.

However it is discouraged from actually doing this as there is no
attempt at API stability at the low level implementation layer.
closed 2020-05-29 by astrogeco