nasa/osal #454: Improve error code on RTEMS OS_stat() implementation

jphickey, 2020-05-13:
**Is your feature request related to a problem? Please describe.**
Depending on the filesystem in use, the RTEMS `statvfs()` call might not be implemented at the filesystem level.  In particular if it is called on the IMFS filesystem type, it returns an error and sets errno to `ENOSYS`.

OSAL translates any error here to `OS_ERROR`, which is what gets returned to the application.

Ultimately this causes the filesystem unit test to fail when this filesystem type is in use.

**Describe the solution you'd like**
Preferable to return `OS_ERR_NOT_IMPLEMENTED` in this case.  In particular, unit tests already check for this, and will skip the test cases for this API, avoiding failure.  This makes it a soft error.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

joelsherrill, 2020-05-13:
Alternatively, you could implement statvfs() for the IMFS. <shameless hint>RTEMS is open source and encourages submissions.</shameless hint> :)

jphickey, 2020-05-13:
> Alternatively, you could implement statvfs() for the IMFS. RTEMS is open source and encourages submissions.</shameless hint> :)

I think maybe the reason its not implemented is because IMFS is dynamically sized, so there isn't a predefined limit for the total size of this FS.

Still, the `ENOSYS` errno should logically translate to `OS_ERR_NOT_IMPLEMENTED` either way, as this is basically what it means.

PR #463 (2020-05-26): Fix #454, Better error translation on statvfs()
**Describe the contribution**
Map the ENOSYS errno to OS_ERR_NOT_IMPLEMENTED, which is more appropriate than OS_ERROR.

Fixes #454

**Testing performed**
Build and execute  unit tests on RTEMS 4.11.3

**Expected behavior changes**
Unit test cases which use OS_statfs and run on an RTEMS IMFS volume (default for the root file system) will be skipped and categorized as "NA" due to OS_ERR_NOT_IMPLEMENTED response, rather than a failure.

**System(s) tested on**
RTEMS 4.11 on pc686 via QEMU

**Contributor Info - All information REQUIRED for considerat [... cut]
commit 45c9742c (2020-05-26): Fix #454, Better error translation on statvfs()

The statvfs() call depends on the underlying filesystem
supporting this.  On RTEMS, the IMFS filesystem type
does not, and it returns the ENOSYS errno.  This is better
translated to OS_ERR_NOT_IMPLEMENTED rather than OS_ERROR.
commit 51ba5aad (2020-05-26): Merge pull request #463 from jphickey/fix-454-statvfs-err-translation

Fix #454, Better error translation on statvfs()